dfki-ric/phobos #159: No module named 'phobos.utils.base'

rathken, 2018-08-29:
Hi, 

Getting the following error. 

phobos-0.8.tar.gz
Ubuntu 16.04.4
blender-2.79b-linux-glibc219-x86_64 
Same error from apt install blender 

which python
/bin/blender-2.79b-linux-glibc219-x86_64/2.79/python/bin/python

python --version
Python 3.5.3

From blender 
PYTHON INTERACTIVE CONSOLE 3.5.3 (default, May 18 2017, 14:40:48)  [GCC 5.4.1 20161019]

workstation:~/bin/blender-2.79b-linux-glibc219-x86_64$ ./blender
Read prefs: /data/nick/.config/blender/2.79/config/userpref.blend
found bundled python: /data/nick/bin/blender-2.79b-linux-glibc219-x86_64/2.79/python
Importing phobos
IMPORT:  phobos.defs
Parsing definitions from: /data/nick/.config/blender/2.79/scripts/addons/phobos/definitions
  defaultJoints.yml
  defaultMaterials.yml
  defaultModel.yml
  defaultMotors.yml
  defaultSensors.yml
  defaultSoftware.yml
  submechanisms.yml
IMPORT:  phobos.io
Registered mesh plugin: ['stl', 'dae', 'obj']
Registered entity plugin: ['generic']
ERROR in entities/__init__: "heightmap.py" has no valid entity plugin interface.
Registered entity plugin: ['light']
Registered entity plugin: ['primitive']
Registered entity plugin: ['smurf']
Registered entity plugin: ['srdf']
Registered entity plugin: ['thumbnails']
Registered entity plugin: ['urdf']
Registered entity plugin: ['yaml']
Registered mesh plugin: ['stl', 'dae', 'obj']
ERROR in scenes/__init__: "mars.py" has no valid scene plugin interface.
Registered scene plugin: ['smurfs']
RELOAD:  phobos.model
IMPORT:  phobos.old_code
IMPORT:  phobos.operators
IMPORT:  phobos.phobosgui
RELOAD:  phobos.phoboslog
IMPORT:  phobos.testing
RELOAD:  phobos.utils
IMPORT:  phobos.utils.base
Traceback (most recent call last):
  File "/data/nick/bin/blender-2.79b-linux-glibc219-x86_64/2.79/scripts/modules/addon_utils.py", line 331, in enable
    mod = __import__(module_name)
  File "/data/nick/.config/blender/2.79/scripts/addons/phobos/__init__.py", line 133, in <module>
    import_submodules(phobos, verbose=True)
  File "/data/nick/.config/blender/2.79/scripts/addons/phobos/__init__.py", line 75, in import_submodules
    results[full_name] = importlib.import_module(full_name)
  File "/data/nick/bin/blender-2.79b-linux-glibc219-x86_64/2.79/python/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
ImportError: No module named 'phobos.utils.base'

Amudtogal, 2018-08-29:
Hi rathken,
Thanks for the bug report.

I just tried to reproduce the error you submitted.

In the whole tar.gz directory there is no single mention of *phobos.utils.base*.
So the error seems to come from an old file which has not been deleted properly.

Could you try the following clean setup:

1. Delete the phobos folder in your Blender config directory (found in `~/.config/blender/2.79/scripts/addons/`
2. Remove the config folder of Phobos **make sure you backup all your custom definitions first** (found in `~/.config/`)
3. Reextract the `tar.gz` file to a new location and rerun `setup.py`

This should fix the problem. Could you come back and tell us whether it worked?

Good luck and happy blending,
Amudtogal

P.S. Optional step: check out our new release 1.0, which will be available by end of the next week...

rathken, 2018-08-29:
Thanks for looking. Very strange, I'll have to take a look and see if there are left over files. I removed ~/.config/blender and did a new install of blender and phobos from the 0.8 release tar.gz and have the same issue. However in a clean newly installed VM it works just fine. I'll have to dig and find where the files seem to think they are hanging out.... clearly some strange install issues on my end. Looking forward to 1.0!

closed 2018-08-29 by rathken