dfki-ric/phobos #167: Defining the kinematic skeleton using parent local frames

ShravanTata, 2018-10-27:
**Is your feature request related to a problem? Please describe.**
Currently to the best of my knowledge I see that each element of the kinematic chain are defined as independent armatures. This makes it very difficult define the kinematic chain global space. It is much more easier to establish a parent-child relationship to describe kinematic chains. 

**Describe the solution you'd like**
To be able to define the kinematics under one Armature with parent-child hierarchy. 

**Additional context**
It is very useful when you do not have a CAD model and want to first describe the kinematics in Blender and then build the visuals and collisions upon it. 

Thank you.

AlCap23, 2018-11-06:
Hi ShravanTata!

Thanks for your input and your interest in phobos!

As I understand it, you would like to be able to do a workspace analysis based upon the kinematic chain and use the "real" armatures / bones as intended via blender for rigging, which would definitely be a nice feature. 

However, this would destroy our current data storage system, which relies on the object in blender to store information such as joint type, effort limits, submechanisms. Likewise we use the pose of each armature object as a frame definition - which is unfortunately not available in single bones.

You can, however, define a simple kinematic structure with just links and joints ( see attached screenshot ).

[image]

Which results in the following urdf:

```xml
<?xml version="1.0"?>
<!-- created with Phobos 0.7 -->
  <robot name="Test">

    <link name="new_link">
    </link>

    <joint name="new_link.000" type="revolute">
      <origin xyz="0.03422 0.46156 -0.10574" rpy="-1.5708 0 0"/>
      <parent link="new_link"/>
      <child link="new_link"/>
      <axis xyz="0 0 1.0"/>
      <limit lower="-0.34907" upper="0.34907" effort="0" velocity="0"/>
    </joint>

    <joint name="new_link.000.001" type="prismatic">
      <origin xyz="0.03422 0.46156 -0.10574" rpy="-1.5708 0 0"/>
      <parent link="new_link"/>
      <child link="new_link"/>
      <axis xyz="0 0 1.0"/>
      <limit lower="-0.34907" upper="0.34907" effort="0" velocity="0"/>
    </joint>

  </robot>
```

I am currently working on the documentation / phobos related stuff on (phobos) friday and hope to get the video tutorials running by the end of next week.

Amudtogal, 2019-02-01:
I guess the questions have been answered.

closed 2019-02-01 by Amudtogal