dfki-ric/phobos #169: Does this plugin support simulation in Blender?

Guptajakala, 2018-11-02:
Hi, Im using the provided model for test. As I turn on animation, the model tear apart...
[image]
How can I keep the robot's joint relationship to prevent the components break down? Is this supported yet? Thanks!

Guptajakala, 2018-11-05:
Anyone has idea? It seems the relationship of joints defined in phobos is different from the rigid body constraints defined Blender?

Amudtogal, 2018-11-05:
Hey Guptajakala,
Phobos has not been designed to **simulate** robot models. Of course it would be convenient to run simulations from within Blender, but this is beyond our scope (especially as there are great simulation options out there: Mars and Gazebo).

Phobos is meant to be a WYSIWYG for robot models, to help with creating and exporting different formats.

So just export your model in one of the formats, put it in Mars/Gazebo and watch it work!

Happy blending,
Amudtogal

closed 2018-11-05 by Amudtogal

Guptajakala, 2018-11-05:
@Amudtogal  Got it. Thanks! Phobos makes my life easier to build models in Blender. I later found Blender could be more suitable to run simulation than Gazebo for me since Blender rendering is very powerful. Hope there could be such addon someday.