dfki-ric/phobos #174: SMURF export flavours

AlCap23, 2018-12-20:
**Is your feature request related to a problem? Please describe.**
SMURF is general and allows a lot of freedom. The parser is defined via the simulator used.
This can produce wrong parsing of objects from the internal phobostype to external definitions, e.g. controllers in MARS.

**Describe the solution you'd like**
On checking the smurf export a menu similar to sdf export should pop up, where the phobostypes to export can be choosen and renamed if necessary. Maybe yaml files can be used to store certain mappings.

**Describe alternatives you've considered**
No ideas right now. Hence, this issue is open for discussion.

Amudtogal, 2019-01-28:
We would like to have a dropdown menu to select the MARS smurf flavour, which would then replace the *controller* files with a different key *motorcontrollers*.
That kind of preset creates a mapping between phobostypes (Smurf export types) and the file to be written.

hwiedPro, 2023-01-23:
Motorcontroller parameters will be included in the motor definition. SMURF flavours are discussed for other purposes and will rather be considered for version >= 2.1

closed 2023-01-23 by hwiedPro