dfki-ric/phobos #182: Add numpy on required python packages

gstavrinos, 2019-04-10:
Numpy is required for phobos to work properly, which is not mentioned in the Installation wiki. Only PYaml is mentioned.

Amudtogal, 2019-04-10:
`numpy` is part of the Blender internal python, so it should be included without any hassle.

If you experience any issue, could you please include the error messages and especially Blender version and OS?

gstavrinos, 2019-04-10:
Hey @Amudtogal, I have no longer access to the error, but was the typical "module not found" python error. I use the recommended blender version and MX Linux 18.1 (essentially Debian stretch). I ran `pip3 install numpy` and the error was solved. Keep in mind that the error occured when I tried to include the add-on through user preferences, and not during setup. After I installed numpy, I rerun setup.py (just to be sure) and after that I could include the add-on normally.

Amudtogal, 2019-04-16:
Hey @gstavrinos,
I just added `numpy` to the dependencies in the installation section.
We had the same issue in #183. Apparently, some Debian versions of Phobos are not compiled with `numpy` included. Thanks for your bug report and happy blending!

closed 2019-04-16 by Amudtogal