dfki-ric/phobos #183: Phobos not activating for Ubuntu 18.04 and Blender 2.79b

bglima, 2019-04-14:
Hi! Thanks for the great work that has been done. It sure helps robotic students like me.

**Describe the bug**
I can't activate Phobos after installing it.  If I try to import it via Blender console, I get the following:
```
PYTHON INTERACTIVE CONSOLE 3.6.7 (default, Oct 22 2018, 11:32:17)  [GCC 8.2.0]
>>> import phobos
Importing phobos
RELOAD:  phobos.defs

Traceback (most recent call last):
  File "/usr/lib/python3.6/code.py", line 91, in runcode
    exec(code, self.locals)
  File "<blender_console>", line 1, in <module>
  File "/home/bglima/.config/blender/2.79/scripts/addons/phobos/__init__.py", line 149, in <module>
    import_submodules(phobos, verbose=True)
  File "/home/bglima/.config/blender/2.79/scripts/addons/phobos/__init__.py", line 67, in import_submodules
    results[full_name] = importlib.reload(modules[full_name])
  File "/usr/lib/python3.6/importlib/__init__.py", line 166, in reload
    _bootstrap._exec(spec, module)
  File "/home/bglima/.config/blender/2.79/scripts/addons/phobos/defs.py", line 32, in <module>
    import phobos.phobossystem as phobossystem
AttributeError: module 'phobos' has no attribute 'phobossystem'
```

I tried reinstalling Phobos after removing ```~/.config/blender/2.79/scripts/addons/phobos``` and ```~/.config/phobos```, but it did not work. 

**Screenshots**
[image]

[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version: 18.04 LTS
 - Blender version: 2.79b
 - Blender Python version: 3.6.
-  Phobos version: release-1.0

bglima, 2019-04-14:
Running ```pip3 install numpy``` solved the problem, without needing to reinstall Phobos again. Maybe related to #182.

Amudtogal, 2019-04-16:
Indeed, it appears `numpy` is not included in the [blender release](https://launchpad.net/~thomas-schiex/+archive/ubuntu/blender) for Ubuntu.
I added the comment for `numpy` requirements to the Wiki.
Thanks for the bug report!

closed 2019-04-16 by Amudtogal