dfki-ric/phobos #184: not appearing in toolbar panel

PakoYam, 2019-05-31:
**Describe the bug**
After installing and enabling, phobos does not appear on the toolbar panel.

1. I have followed the installation manual from https://github.com/dfki-ric/phobos/wiki/Installation

2. After running the code "py -3 setup.py", it didn't ask me anything about my Blender version. 

3. I opened Blender, the add-on is already enabled automatically in User Preferences. But nothing appears on the toolvar panel (see Screenshot 1).

4. If I disable it and try to enable it again, it returns an error. (see Screenshot 2).

**Screenshots**
Screenshot 1:
[image]

Screenshot 2:
[image]

**Desktop (please complete the following information):**
 - Windows 10
 - Blender version: 2.79b
 - Blender Python version 3.5.3

Thank you.

Amudtogal, 2019-06-04:
Hi PakoYam,
at first, thanks for your bug report!

Let us try some steps first:
- the main issue is the missing yaml installation (that actually caused your problem). Could you check whether `yaml` can be imported from your python console (search for python in your windows searchbar and either start *IDLE Python* or the *Python Interpreter*).
 - it is surprising, that Phobos was already enabled in your settings. Did you have Phobos on your computer before? Did you use the `--startup-preset` option when installing? Anyway, try removing the phobos folder from your **HOMEFOLDER**`/AppData/Roaming/Blender Foundation/Blender/2.79/scripts` path (you might need to enable hidden files to see it). Then remove the `python_dist_packages.conf` from your Phobos download folder. After this, Phobos should be removed when starting Blender. A subsequent reinstall might fix the issue...

Can you check these steps?
Best,
Amudtogal

Amudtogal, 2020-01-12:
I assume this was resolved, otherwise please reopen...

closed 2020-01-12 by Amudtogal