dfki-ric/phobos #188: Option to avoid automatic fixing of inertias

ShravanTata, 2019-07-17:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

hwiedDFKI, 2022-01-24:
Hi @ShravanTata 
Thanks for your issue. Do you mean if you have an inertia tensor with values below 1e-3 they are overriden?
What would be an inertia tensor, that you need?
The check is currently there to ensure that the inertia tensor is positive-definite.

hwiedPro, 2023-03-13:
Solved by ef670a4