dfki-ric/phobos #190: The hud_pos property in the camera sensor definition YML file should be hud_idx.

georgeslabreche, 2019-07-24:
**Describe the bug**
The `hud_pos` property in the camera sensor definitions yml file should be `hud_idx`. The former is not recognized by MARS but the latter is.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a model with at least 2 camera sensors.
2. Assign different values for hud_pos.
3. Export the model and load it into MARS.
4. All camera sensor huds are layered on top of each other at hud location index 0.

**Expected behavior**
I expect the huds to be positioned in different location as per their hud_pos values.

**Screenshots**
Not applicable.

**Desktop (please complete the following information):**
- OS: Ubuntu
- Version: 18.04
- Blender version: 2.79b
- Blender Python version: 3.6.7

**Additional context**
The `hud_pos` property is also used for the scanning sonar sensor. Maybe this also needs to be updated: 
https://github.com/dfki-ric/phobos/blob/63682bd6b34d3d6fe9cd357aba5d5e0250dd3b72/config/definitions/defaultSensors.yml#L127

georgeslabreche, 2019-07-24:
I have created a pull request to resolve this issue:
https://github.com/dfki-ric/phobos/pull/191

I'm unaware why that property would have been set to hud_pos instead of hud_idx in the first place. Is there an environment other than MARS where hud_pos is required?

georgeslabreche, 2019-12-09:
@kavonszadkowski just a note that this is a blocker for us and we would like to come up with a way forward.

hwiedPro, 2022-06-16:
This seems to be solved. See https://github.com/dfki-ric/phobos/pull/191#issuecomment-1046871930
Therefore closing this.

closed 2022-06-16 by hwiedPro