dfki-ric/phobos #192: Error in installation wiki

oscar-lima, 2019-07-25:
Following [installation wiki](https://github.com/dfki-ric/phobos/wiki/Installation) on ubuntu 18.04, specifically the execution of the command:

```sudo add-apt-repository ppa:thomas-schiex/blender```

ends up installing blender 2.8, instead of version 2.79 which is the one required by phobos

oscar-lima, 2019-07-25:
The instructions [here](https://www.askmetutorials.com/2017/09/install-blender-279-on-ubuntu-linuxmint.html) worked for me with a slight modification (not working with wget for some reason...):

1. open internet browser, go to https://www.blender.org/download/Blender2.79/blender-2.79-linux-glibc219-x86_64.tar.bz2

and download file

2. execute:

```tar -xvf blender-2.79-linux-glibc219-x86_64.tar.bz2 && cd blender-2.79-linux-glibc219-x86_64```

3. open blender

```./blender```

Amudtogal, 2019-12-17:
This tip was already included in the wiki.

Also, the installation changed a bit, so this should be done!

closed 2019-12-17 by Amudtogal