dfki-ric/phobos #194: Support for Blender 2.8

Xia0ben, 2019-08-23:
**Is your feature request related to a problem? Please describe.**
Blender 2.8 is now out of Beta, and it would be nice to have Phobos work on this most recent version which brings tremendous improvements to the UI, making it far more accessible than before. But the wiki mentions "Blender 2.8 is expected to include major changes that will not be compatible with Phobos." : is Phobos dependent on the Blender Game Engine ? If not, why do you think compatibility would break ?

**Describe the solution you'd like**
Phobos running on Blender 2.8 would be great ! =)

**Describe alternatives you've considered**
A list of blocking points on which people could contribute to fix the hurdle would already be a very good start. ;-)

**Additional context**
If the BGE is indeed the problem, [Armory3D](https://github.com/armory3d/armory) may be of help ?

Anyway, thanks for your great software !

kavonszadkowski, 2019-08-23:
Thanks for the suggestion! Obviously, we've been keeping an eye on Blender 2.8 ever since it was first announced. The problem is not the Blender Game Engine, but the changes in the scripting API on which Phobos is built, see [Blender 2.8 release notes](https://wiki.blender.org/wiki/Reference/Release_Notes/2.80/Python_API).

There has been only very preliminary work on framing the extent of problems we'd run into with Blender 2.8. At the end of the day it's a question of resources to rewrite much of the code-base and whether it's worth to make the transition.

Xia0ben, 2019-08-24:
Thank you for the very fast answer !
Well, if the scripting API changed, it would indeed take quite a bit of time to figure out how to proceed. Is the project page https://github.com/dfki-ric/phobos/projects/9 the preliminary work on determining the possible problems you mentioned ?

ShravanTata, 2019-08-26:
It would definitely be useful to port to blender 2.8 as it is a major release and would love to work with Phobos while making use of the latest features from 2.8. I can help as Phobos has been a useful tool for me.

kavonszadkowski, 2019-08-28:
That's right, @Xia0ben, @AlCap23 has been looking into Blender 2.8 while at the same time reorganizing Phobos, but there's nothing definitive there, yet. I will get back to working on Phobos next week and check how much would have to be done.
Thanks for offering your help, @ShravanTata - much appreciated!

ShravanTata, 2019-09-24:
@kavonszadkowski : Any update on the estimation of porting to blender 2.8?

kavonszadkowski, 2019-11-26:
Currently we have almost no resources to develop Phobos further, so nothing is planned in that respect.

pauceano, 2020-09-16:
Hello,
Any news on the port to Blender 2.8? Maybe I could help if there is some documentation...

AlCap23, 2020-09-28:
Hey! I think the development is currently stopped. Neither @kavonszadkowski nor me are working on Phobos at work. So sadly, there is no progress so far. Sorry :(

malter, 2020-10-05:
Hey,
I will check next week what have to be done for the port. And I will setup a documentation for it.

malter, 2020-10-07:
The core functionality of Phobos should work now on the branch `blender2.9`. I created a file `TODO.md` with the left work and known issues so far. Due to changes in Blender you will find the Phobos panels now on the right hand side of the 3D View. I will continue the work to provide a stable and usable version of Phobos soon.

Xia0ben, 2020-10-29:
Thank you @malter for the work you've done so far !

ManifoldFR, 2020-11-03:
Hi,

I just tried to install phobos from the blender2.9 branch with the latest version of Blender. When I activate the add-on I get
```bash
ERROR in scenes/__init__: "mars.py" has no valid scene plugin interface.
Registered scene plugin: ['smurfs']
[1]    71691 abort (core dumped)  blender/blender
```

malter, 2020-11-03:
The mars.py error is misleading and shouldn't be responsible for the abort. So far I only tested on MacOS. I will check it on ubuntu soon. Do you use the system blender version?

malter, 2020-11-03:
On elementary OS 5.1 (based on Ubuntu 18.04) it works as expected. I installed blender via download from the blender website. Unpacked it in it's own folder. Cloned the phobos project and switched to the branch `blender2.9` and executed the `setup.py` with: `python3 setup.py`. Afterwards, I was able to enable phobos in blender and the UI was updated with the phobs tabs. However, I also have the `mars.py` error but it is ignored and not influencing the plugin in general.

malter, 2020-11-03:
By the way, the basic `Modelling Walkthrough` is now working fine in Blender 2.9. Whereby, I still haven't tested all features of phobos.

ManifoldFR, 2020-11-03:
> The mars.py error is misleading and shouldn't be responsible for the abort. So far I only tested on MacOS. I will check it on ubuntu soon. Do you use the system blender version?

I'm on Ubuntu 20.10 (Groovy) using the tarball version of Blender (straight from the website).

malter, 2020-11-03:
Can you maybe delete the installation.conf in phobos and post the output of `python3 setup.py`?

ManifoldFR, 2020-11-03:
Here it is:
```
Where is Blender installed to? ~/blender/  
Installing PIP...
Looking in links: /tmp/tmp3d1ls6fw
Requirement already satisfied: setuptools in /home/manifold/.local/lib/python3.7/site-packages (41.2.0)
Requirement already satisfied: pip in /home/manifold/.local/lib/python3.7/site-packages (19.2.3)
Collecting pip
  Using cached https://files.pythonhosted.org/packages/cb/28/91f26bd088ce8e22169032100d4260614fc3da435025ff389ef1d396a433/pip-20.2.4-py2.py3-none-any.whl
Installing collected packages: pip
  Found existing installation: pip 19.2.3
    Uninstalling pip-19.2.3:
      Successfully uninstalled pip-19.2.3
Successfully installed pip-20.2.4
... successful.

Installing YAML...
Requirement already satisfied: PyYaml in /home/manifold/.local/lib/python3.7/site-packages (5.3.1)
Requirement already up-to-date: PyYaml in /home/manifold/.local/lib/python3.7/site-packages (5.3.1)
... successful.

Removed older installation.
Copied addon files to /home/manifold/.config/blender/2.90/scripts/addons/phobos.
Installation successful!
```

malter, 2020-11-03:
Looks not that wrong. :-(
So, I don't have an information what could go wrong. Maybe you can post the content of `/home/manifold/.config/blender/2.90/scripts/addons/phobos/installation.conf`. But I assume that will also look fine.
Do you start blender in the terminal? And you get the error if you try to activate the phobos addon?

kavonszadkowski, 2020-11-03:
I'm currently looking into getting rid of the installation process the way it works at the moment. If you're lucky I could merge that tomorrow. Hopefully should take care of those problems

malter, 2020-11-03:
Ok, after comparing again with my setup, the problem seems to be, that py-yaml is installed into the system python instead of the blender python folder. Maybe, some environment variable produces that behavior. You can try to manually copy:
`/home/manifold/.local/lib/python3.7/site-packages/yaml` to `/home/manifold/blender/2.90/pyhton/lib/python3.7/site-packages`. If this works, at least we know where the error comes from. ;-)

ManifoldFR, 2020-11-03:
> Looks not that wrong. :-(
> So, I don't have an information what could go wrong. Maybe you can post the content of `/home/manifold/.config/blender/2.90/scripts/addons/phobos/installation.conf`. But I [...]

Here is the output, looks quite normal to me:
```
/home/manifold/blender/2.90/python/lib/python3.7/site-packages # Python package installation path
/home/manifold/blender/ # Blender installation path
/home/manifold/blender/2.90/python/bin/python3.7m # python executable
/home/manifold/blender/blender # Blender executable
3.7 # Python version
2.90 # Blender version
```
And I fixed the PyYAML install path. I'm launching from the command line. I tried doing `conda deactivate` before, also.

malter, 2020-11-03:
So, after manually copying the yaml folder into the blender site-packages it is still not working?
What's the output of the setup.py after deactivating conda? Unfortunately I have no experience with conda. You could also post the output of `export` after deactivating conda. Maybe we get a hint there.
If nothing helps, either I have to reproduce your setup in a virtual machine or Kai can provide a better alternative to install python dependencies into blender.

ManifoldFR, 2020-11-03:
I was able to enable the plugin just now after launching Blender using the NVIDIA dedicated GPU. I don't know why that was the reason.

Explanation of my setup (not an unusual one):

- I am on a Dell XPS 15 laptop with dual graphics: Intel UHD 630 (integrated chipset running Mesa drivers I think) & NVIDIA GTX1650 (using NVIDIA proprietary drivers)
- I was launching Blender without GPU enabled - it was running on the Intel chipset
- Launching Blender with GPU fixed it

kavonszadkowski, 2020-11-04:
So I looked into this issue today and turns out it's not as easy to solve as I thought. Turns out not all distributions' Blender python bundles come with pip, which would have made it easy to install additionally required libraries.
There may be another way around this by getting rid of those dependencies in the first place, which is what I'm looking into now.
Thanks for your patience!

malter, 2020-11-05:
> I was able to enable the plugin just now after launching Blender using the NVIDIA dedicated GPU. I don't know why that was the reason.
> 
> Explanation of my setup (not an unusual one):
> 
> * I am on a Dell XPS 15 laptop with dual graphics: Intel UHD 630 (integrated chipset running Mesa drivers) & NVIDIA GTX1650
> * I am using Ubuntu 20.10 and NVIDIA 455.28 proprietary drivers
> * I was launching Blender without GPU enabled - it was running on the Intel chipset
> * Launching Blender with GPU fixed it
> 
> It might be related to some issue upstream: https://developer.blender.org/T76384

Hey, great to hear that it is working now! It would be interesting to know if only the phobos add-on produces the problem or if blender also crashes when enabling other add-ons. Thank you for your investigations!

malter, 2021-12-15:
Finally, blender 2.9 (also blender 3.0) is supported and merged to the master branch. We will give it a test phase and fix last bugs to then create a new release.

closed 2021-12-15 by malter