dfki-ric/phobos #195: Support for multiple degrees of freedom joints

ShravanTata, 2019-08-28:
Currently Phobos only supports single degrees of freedom joints. In my models it is essential to have multiple degrees of freedom. From a robotics points of view it is understandable to have single DOF since they represent motors. But I am working on more abstract simulation models which would require multi DOF joints.

**Describe the solution you'd like**
To support all the joints described in SDF format. 
[http://sdformat.org/spec?ver=1.6&elem=joint]

cjaverliat, 2020-11-05:
I encountered the same issue regarding ball joints.
As of now, there are three joints missing for the SDF format (Cf. [joints section](http://sdformat.org/tutorials?tut=spec_model_kinematics&cat=specification&#joint)):
- ball: 3 rotational degrees of freedom
- screw: 1 coupled rotational/translational degree of freedom
- universal: 2 rotational degrees of freedom
As Phobos is mainly based on URDF it explains the lack of such joints.

**What would be the ideal solution in that case ?**
- On the one hand, adding all the SDF joints types would resolve the issue, but on the other they would be useless and not exported for the URDF format. 
- On the other hand, it seems that you can build up a ball joint/spherical joint by composing fixed and continuous joints (Cf. [this discussion](https://answers.ros.org/question/258420/ball-joint-in-urdf/)). Maybe it would be possible to convert automatically the ball joint into the four equivalent joints in case of a URDF export.
- Do not change anything and let the user create its own ball joint the way I mentionned before, but adding it somewhere in the documentation for the kinematic skeleton.

hwiedPro, 2023-01-23:
Adding a ball joint is no problem, exporting it to URDF will lead to an error message, so that the user has to take care of it. I'll update this when I found time to tackle this.

The other two need some more changes, especially the coupling of the screw joint. This will rather happen in a later iteration.