dfki-ric/phobos #196: Phobos not activating for macOS Sierra 10.12.6 and Blender 2.79b

zengtaiping, 2019-09-01:
**Describe the bug**
I can't activate Phobos after installing it. The same error in #183 .
```
AttributeError: module 'phobos' has no attribute 'phobossystem'
```
I tried reinstalling Phobos and `pip3 install numpy`, but it did not work.

**Screenshots**
[image]

**Desktop:**
 - OS: macOS
 - Version Sierra 10.12.6
 - Blender version 2.79b
 - Blender Python version 3.7.4

Thanks in advance!

ShravanTata, 2019-09-02:
I had the same issue but then once I restarted blender it was fine. But then I ran into the error of bUtils.getPhobosConfigPath().modelsfolder was returning a str instead of an object.

If fixed it with the following changes in phobos/libraries/models.py
Note that it's a temporary hack!

```
def compileModelList():
    """TODO Missing documentation"""
    from bpy.props import EnumProperty
    from bpy.types import WindowManager

    # DOCU missing some docstring
    log("Compiling model list from local library...", "INFO")

    # clear old preview collections
    for previews in model_previews.values():
        bpy.utils.previews.remove(previews)
    model_previews.clear()
    model_data.clear()

    try:
        rootpath = bUtils.getPhobosConfigPath().modelsfolder
    except KeyError:
        log('Can not create mechanism preview. Phobos not registered.', 'DEBUG')
        return
    + except AttributeError:
    +   rootpath = ''
    +   log('Rootpath not found!', 'DEBUG')
    +   return

    if rootpath == '' or not os.path.exists(rootpath):
        log('Model library folder does not exist.')
        return
```

zengtaiping, 2019-09-03:
@ShravanTata Thanks so much for your reply and your kind sharing. 
After I restarted the blender and my computer many times, it still did not work. Later, I will try it on Ubuntu 18.04, and see whether it would work fine.

closed 2019-09-09 by zengtaiping

mnbf9rca, 2019-10-08:
On my machine, the default (OS) Python is 3.7 but phobos is using 3.5. So I had to:
- create a new env with conda with 3.5 so i could run the phobos setup script
- pip3 install PyYaml
- copy `PyYAML-5.1.2.dist-info` and `yaml` from `/usr/local/lib/python3.7/dist-packages/` to /`usr/local/lib/python3.5/dist-packages/`