dfki-ric/phobos #197: Canno't see Phobos in the addons

asawaswapnil, 2019-09-07:
I have linux 18.04. Both blender2.79 python is 3.6.8 and system python is in 3.6.8
Issue: I can't find phobos in the addons after taking the required steps. I even installed and installed blender again at the beginning. 
Here are the commands that I used:

root@swapnil-Lenovo-ideapad-530S-15IKB:/home/swapnil/phobos#  sudo apt-get install blender
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  libccd-dev libccd2 libllvm7 libllvm7:i386 libopenal-dev
  libqtpropertybrowser4
Use 'sudo apt autoremove' to remove them.
The following NEW packages will be installed:
  blender
0 upgraded, 1 newly installed, 0 to remove and 1 not upgraded.
Need to get 26.6 MB of archives.
After this operation, 109 MB of additional disk space will be used.
Get:1 http://us.archive.ubuntu.com/ubuntu bionic-updates/universe amd64 blender amd64 2.79.b+dfsg0-1ubuntu1.18.04.1 [26.6 MB]
Fetched 26.6 MB in 2s (10.7 MB/s)   
Selecting previously unselected package blender.
(Reading database ... 302770 files and directories currently installed.)
Preparing to unpack .../blender_2.79.b+dfsg0-1ubuntu1.18.04.1_amd64.deb ...
Unpacking blender (2.79.b+dfsg0-1ubuntu1.18.04.1) ...
Setting up blender (2.79.b+dfsg0-1ubuntu1.18.04.1) ...
Processing triggers for mime-support (3.60ubuntu1) ...
Processing triggers for desktop-file-utils (0.23-1ubuntu3.18.04.2) ...
Processing triggers for gnome-menus (3.13.3-11ubuntu1.1) ...

root@swapnil-Lenovo-ideapad-530S-15IKB:/home/swapnil/phobos# pip3 install PyYaml
Requirement already satisfied: PyYaml in /usr/lib/python3/dist-packages

root@swapnil-Lenovo-ideapad-530S-15IKB:/home/swapnil/phobos# pip3 install numpy
Requirement already satisfied: numpy in /usr/local/lib/python3.6/dist-packages

root@swapnil-Lenovo-ideapad-530S-15IKB:/home/swapnil# git clone https://github.com/dfki-ric/phobos.git
Cloning into 'phobos'...
remote: Enumerating objects: 61, done.
remote: Counting objects: 100% (61/61), done.
remote: Compressing objects: 100% (42/42), done.
remote: Total 11888 (delta 24), reused 45 (delta 19), pack-reused 11827
Receiving objects: 100% (11888/11888), 12.44 MiB | 7.73 MiB/s, done.
Resolving deltas: 100% (8678/8678), done.

root@swapnil-Lenovo-ideapad-530S-15IKB:/home/swapnil# cd phobos/

root@swapnil-Lenovo-ideapad-530S-15IKB:/home/swapnil/phobos# git checkout release-1.0
Branch 'release-1.0' set up to track remote branch 'release-1.0' from 'origin'.
Switched to a new branch 'release-1.0'

root@swapnil-Lenovo-ideapad-530S-15IKB:/home/swapnil/phobos# python3 setup.py
You started this script with python version: 3.6
Make sure Blender uses the same version!

root@swapnil-Lenovo-ideapad-530S-15IKB:/home/swapnil/phobos# python3 setup.py --startup-preset
You started this script with python version: 3.6
Make sure Blender uses the same version!
python_dist_packages.conf found! Configuration done.

root@swapnil-Lenovo-ideapad-530S-15IKB:/home/swapnil/phobos# python3 --version
Python 3.6.8

[image]

kavonszadkowski, 2019-09-11:
First of all, to check if Phobos was installed properly you could check if the following folder exists and is populated with Phobos:
``~/.config/blender/2.79/scripts/addons/phobos``

Amudtogal, 2019-12-17:
Hi,
would you mind checking the problem with the current release 1.0.1?

This might have resolved your issue...

Amudtogal, 2020-01-12:
I assume this was fixed, otherwise please reopen...

closed 2020-01-12 by Amudtogal

Omar2664, 2022-08-15:
my phobos exist 
~/.config/blender/2.82/scripts/addons/phobos

but it's not showing on blender addons 
[image]

malter, 2022-08-22:
How have you installed Phobos? Did you tried the released zip file: https://github.com/dfki-ric/phobos/releases/download/1.0.3/phobos.zip and installed it via the install button in Blender Preferences Addons?

esamet, 2023-07-19:
> [https://github.com/dfki-ric/phobos/releases/download/1.0.3](https://github.com/dfki-ric/phobos/releases/download/1.0.3/phobos.zip)

finally it soved my issue, when you clone normally from github it wouldnt work