dfki-ric/phobos #198: Can't activate phobos addon in Linux Deepin os.

Sumon-miazi, 2019-11-02:
I follow the procedure which is given in the installation wiki [https://github.com/dfki-ric/phobos/wiki/Installation](url) . And I am using deepin os and have python3.5.

and my addon page shows this error informations:-
[image]

and I downloaded the latest release version but my version number is pointing to 0.7.1. Is this ok?
[image]

How can I solve this? help, please...

Amudtogal, 2019-12-17:
Could you retry on the new release?

We changed the installation, so this might have resolved your issue...

Amudtogal, 2020-01-12:
I assume this is resolved, otherwise please reopen...

closed 2020-01-12 by Amudtogal