dfki-ric/phobos #202: Importing urdf in python scripting

EdwardSmith1884, 2020-01-17:
Hi,

is it possible using this package to import urdfs via python scripting directly? It looks like your importUrdf function returns a dictionary representing the model, but is it possible to pass this to blender in a python script? 

Thanks!

Amudtogal, 2020-01-18:
Hi Edward,

what exactly is it you want to achieve?
Our internal data representation when loading or saving a model is indeed a dictionary. However, this dictionary is then transformed into a Blender representation (armatures, meshes etc). All of this happens when you use the "Import Robot Model" button in the toolbar. This should leave you with a working model in Blender.

If you want to dive into Blender scripting and use this function for your own script, it gets a bit more complicated. The relevant functionality can be found in the "bpy.ops.phobos.import_robot_model()" operator.

Could you be a bit more specific with your question?

Best,
Simon

EdwardSmith1884, 2020-01-18:
Hi Simon, 

What I want to achieve is being able to load urdf files through a python script so that I can perform scripted rendering on arbitrary robot inputs. Looking into your code a little I saw that your urdf function parses the urdf into a dictionary, and I wanted to know if how to load this dictionary into blender through python. From what  you wrote this can be achived via "bpy.ops.phobos.import_robot_model()"? 

Thanks!

Amudtogal, 2020-01-19:
Hi Edward,

The index of the Phobos API can be found [here](https://dfki-ric.github.io/phobos/genindex.html).

The function you were looking for is the [buildModelFromDictionary](https://dfki-ric.github.io/phobos/phobos.model.html#phobos.model.models.buildModelFromDictionary) function, which takes a dictionary (as imported before) and creates a model representation in Blender.

However, be careful with using these functions. Lots of the requirements are covered in the respective operator (which is the class that creates a button in the Blender GUI, eg `bpy.ops.phobos.import_robot_model()`).

I think this should be all you need to get your script started!
If it is a usefull addition to Phobos, consider a pull request :)

Otherwise, please give me a short reply, whether I can close this issue...

EdwardSmith1884, 2020-01-21:
Thank you this was very helpful, the issue can be closed.

closed 2020-01-21 by EdwardSmith1884

yenchenlin, 2021-02-15:
Hello @EdwardSmith1884 , could you explain how you make things work?