dfki-ric/phobos #206: Installation on MacOS

artificial-cassiano, 2020-02-26:
Hi, 
thanks for this super useful plugin!

When I try to install on MacOS (OSX Catalina) by running:
`python3 setup.py --blender /Applications/`
I get this following error message: 
`Could not identify the blender subfolder.
Installation aborted!`

on Mac Blender (2.8) is installed as an app on `/Applications/Blender.app`, not a directory. How should I pass this argument to setup.py?

thanks

malter, 2020-02-27:
Hey,

I have a open pull request which includes some notes for installing Phobos on MacOS:
https://github.com/dfki-ric/phobos/pull/204/commits/4d64775bb4a49fcc2b2e922456df884a92be80c1

By the way, Phobos is currently not compatible with Blender 2.8, to use Phobos you should also install Blender 2.79.

Amudtogal, 2020-02-27:
I will work on this issue (and the pull-request) this weekend!

Currently on a conference :)

Amudtogal, 2020-03-02:
I have updated (but not yet pushed) the setup.py to work on macOS as well.
Right now, I am hunting down someone around here, to test it before I push.
Expect it this afternoon...

Amudtogal, 2020-03-02:
Could you check the new installation, and in particular the [macOS section](https://github.com/dfki-ric/phobos/wiki/Installation#macos-1) in the wiki?

Before reinstalling, please remove the installation.conf file first and make sure you installed Blender 2.79, as described in the wiki.

Regards

Amudtogal, 2020-03-23:
I assume this is fixed, otherwise reopen.

closed 2020-03-23 by Amudtogal