dfki-ric/phobos #207: Import world sdf

eliabntt, 2020-04-07:
Good morning,
I've a question for you. How is the development going on to import a world sdf file into blender? 

I'm trying to find a way to import it because my world is made up of different dae files organized together but I need a single one of them describing the whole world (even dae or other 3D formats obj...). 

Any tip?

Amudtogal, 2020-04-10:
Hi Elia,
so far, I haven't continued working on the SDF world import.
I am working on a tutorial series at the moment, and I can't promise to work on this in the next month...

Until then, you could try to import your features individually into Phobos. The SDF export is working so far, you might give it a try.

Best,
Simon

eliabntt, 2020-04-14:
Yup ok but I don't want to re-import manually everything and place everything in the same place again otherwise I could use blender directly. I've found these environments online (3DGems) and they're made of many dae/meshes. But, unlike most environments (like kitchen_dining) where the ".world" files include a SINGLE .dae describing the whole world they're ".world" file links and position these many different meshes. I'm trying to find a way to combine them in one single big file but I'm starting to think this is quite difficult :)

Amudtogal, 2020-04-15:
Hey,

could you send me a reference file to have a look at?
I might find some time this week to look into this...

eliabntt, 2020-04-15:
So [this](http://data.nvision2.eecs.yorku.ca/3DGEMS/) is what I have. As you can see from the world's archive each .world file within is made of various models(dae, stl..). 
Instead [here](https://bitbucket.org/osrf/gazebo_models/src/default/kitchen_dining/model.sdf) you can see how the sdf has a single dae file describing the whole world.

Amudtogal, 2020-04-21:
Hi,
looking into the kitchen file you provided: you can import it into Phobos (there is a small bug concerning the empty object which groups the visual objects inside the collada file, but you can ignore it).

Exporting is a bit more difficult. We would need to allow Empty objects in Blender to work to group meshes together. This is definitely a useful feature, but is not changed easily. I will work on it, but for the time being consider this:

- import your files, attach them to an empty object within blender (parenting Ctrl p)
- manually export that empty as a collada file and tick the `children` box in the export settings
- you should have a single collada file, which contains all your meshes
- manually add the collada file to your world sdf

This should give you a working model, which can be imported into Phobos, although the export will not group your meshes...

Amudtogal, 2020-04-21:
I am working on a meshgroup import/export, which allows to group visual/collision objects underneath an empty (which is what collada does for your kitchen file), but this will take a bit...

Progress is here:
/dfki-ric/phobos/tree/dev/meshgroupsIO

eliabntt, 2020-04-22:
But for the kitchen world file it's unnecessary work. By being a world file with a _single_ .dae (collada) file you can directly import that into blender and have the whole world. Ok, you can't separate objects but it's not what I want.. What I want to do is to obtain a single representation from the others world files (The ones of the first link). These specify for each collada file it's position and orientation, instead I would like a _single_ collada/obj/whatever file representing my whole world without having to re-place manually all the objects. Am I making myself clearer?

closed 2021-03-24 by eliabntt