dfki-ric/phobos #208: Error while trying to create a joint

ahgartner, 2020-05-18:
Hello,

I was going through the tutorial to add kinematics to my model and then when I tried to create a joint, this error occured and my model disappears. Only after I click a node, my model appears again.
Could you please help me?
[image]

I am working on Windows 10, Blender 2.79b.

Thanks in advance!

kavonszadkowski, 2020-06-04:
Most of the work on and with Phobos has been done on Linux and this bug seems to stem from the assumption that the resource files are always in the same place. Which I suspect they are not under Windows. I'll have to look into it.

hwiedDFKI, 2022-06-16:
This could be solved by blenders bpy.utils.user_resource.
@kavonszadkowski Do you have updates?

malter, 2022-06-16:
I think we should try to reproduce the bug in the current master version. Maybe with the current Blender version on Windows it is already solved. I can try that out next week.

hwiedPro, 2023-01-27:
Couldn't reproduce in the current version. Please reopen if this persists for you.

closed 2023-01-27 by hwiedPro