dfki-ric/phobos #213: Naming problem for automatically created/imported objects

bosforox, 2020-09-28:
**Describe the bug**
With Blender-phobos co-work, strange naming problems occur when objects are created or imported with same names. This issue causes some problems for further steps in the creation of a composite model(Q-Rock atomic model import). If a name already exists in Blender, a new object with the same name is created with a subfix of .00x, which in a recursion can lead to a chain of possible subfixes like obj_name.001.001.001.001. Maybe the reuse of dots can be avoided or the names can be taken either from parent objects or from prefix (link_.. , visual_... etc). 

@2maz

malter, 2020-10-05:
This is not really a bug since in Blender the object name is the unique identifier and blender does this name handling on purpose. Normally, Blender identifies .00x and uses .00[x+1] as new name instead of creating chains.
Recursive model imports are not part of Phobos either.
So, what can Phobos do for you to improve your application?

hwiedDFKI, 2022-06-16:
I consider this as solved, due to inactivity

closed 2022-06-16 by hwiedDFKI