dfki-ric/phobos #216: Support upbge - add support for BT_Multi_body()

BluePrintRandom, 2020-10-29:
**Is your feature request related to a problem? Please describe.**
upbge has bullet physics engine - but BT_MultiBody is not currently wrapped 

**Describe the solution you'd like**
wrap BT_Multibody in upbge and list it as a target

**Additional context**
https://www.youtube.com/watch?v=XeeYf1HHxeY - UPBGE is a robust physics environment coupled with a realtime python api

hwiedDFKI, 2022-06-16:
Could you please elaborate on what exact feature/ability you expect us to add?

But according to your video: Phobos is a tool to annotate Robot models exported from CAD to prepare it for simulation, not to simulate/animate the model directly in Blender. Therefore I label it for the time being as wontfix. If I got you wrong and you elaborate your request further I'll update.

hwiedPro, 2023-01-27:
This seems to be stale. Please reopen otherwise

closed 2023-01-27 by hwiedPro