dfki-ric/phobos #217: How to create the relationship of two rotating gears

mn-banjar, 2020-10-30:
Hi, I'm trying to use this plugin to create urdf file of a robot arm.
How to configure two rotating gears?

malter, 2020-11-02:
Hi, I'm sorry I didn't get the question right away. The [Modeling Walkthroug](https://github.com/dfki-ric/phobos/wiki/Modeling-Walkthrough) describes the way you can model a robot arm in general. But what kind of configuration do you mean with two rotating gears? Normally, for each driven rotating axis you create a joint and a motor in urdf. You can create joints without motors for passive DOFs and you can use mimic joints to clone movements between joints.

mn-banjar, 2020-11-02:
> Hi, I'm sorry I didn't get the question right away. The [Modeling Walkthroug](https://github.com/dfki-ric/phobos/wiki/Modeling-Walkthrough) describes the way you can model a robot arm in general. [...]

Thank you I solved the issue
yes, It was about using mimic joints
Actually, I wrote the tag by hand on the urdf file

closed 2020-11-02 by mn-banjar