dfki-ric/phobos #222: Phobos installation fails

lucwens, 2021-02-15:
I try to install phobos on windows 10, but the setup does not get very far.
Blender 2.79b is installed in d:\Blender, the executable is at D:\Blender\blender.exe.
Phobos is in d:\Phobos.
My steps:
- I open a command window with administrator rights
- d:
- cd \phobos
- "d:\Blender\2.79\python\bin\python" setup.py
-  >> Where is blender installed to?
- d:\blender   [ right? blender is actually in this directory]
- >> Could not find Blender executable
- >> Installation aborted

I tried all kinds of variations, nothing,.....sigh

Luc

lucwens, 2021-02-16:
I got it working by changing line 99 in setup.py from 
blender_executable = path.join(blender_path, 'blender')
to 
blender_executable = path.join(blender_path, 'blender.exe')

I don't have the impression that this setup.py has ever been tested on windows...

malter, 2021-04-27:
I redesigned the install process for the new blender version, which still have to be tested but looks promising: https://github.com/dfki-ric/phobos/tree/blender2.9
I will close the issue, since the old setup.py will become deprecated soon. ;-)

closed 2021-04-27 by malter