dfki-ric/phobos #223: Not able to import and work with stl files

gautam-sh, 2021-03-24:
**Describe the bug**
I am able to import stl file but then it is imported as a single element. Hence I am not able to select different components as different links to proceed further. And it gets killed automatically after some time.

**To Reproduce**
Steps to reproduce the behavior:
Import stl file.

**Expected behavior**
Model for editing and creating urdf file.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version 18.04

While adding the addon I also opted for the option to modify default blender settings as mentioned in the installation instructions.

malter, 2021-03-24:
Just to get it right: You import a single STL file and then want to model your robot by adding links, collisions, etc.?
And in your case the STL import of Blender creates a single object?
If I got it right, in the first place you have two options:
  1. Check the source of the STL and if it is possible to change the export option, like exporting single STL for single components
  2. If that is not possible, you have to switch into Blender "edit-mode" and separate the objects manually. Depending on the model this can require more or less effort. In many cases you can select a single vertex in edit-mode and type "Ctrl-L" to select all linked vertices which represent one complete components, if you are lucky.

Hope that helps a bit.

Concerning the crash I didn't get when it is happening. Maybe Blender becomes unstable if the model is too large. Or does it happen while you where working with the Phobos tools?

malter, 2021-06-24:
Closed due to no feedback

closed 2021-06-24 by malter