dfki-ric/phobos #225: Cannot export joint damping to URDF

Spike87, 2021-04-29:
**Describe the bug**
When setting "damping constant" when defining a joint, this property does not show up in the URDF tags of the joint.

**To Reproduce**
Steps to reproduce the behavior:
1. Select joint (right click)
2. Click on 'Define joint(s)'
3. Set 'damping constant' to 0.2
4. Select robot hierarchy
5. Click on 'Export model'
6. No 'dynamics damping' tag shows in the URDF file

**Expected behavior**
When setting "damping constant" when defining a joint, the 'dynamics damping' tag should show up in the URDF file.

**Desktop (please complete the following information):**
 - OS: Ubuntu
 - Version 20.04.2 LTS
- Blender version: 2.79b
- Phobos version: 1.0.1

hwiedDFKI, 2022-06-16:
We are aiming to have this included in the next major release on which we are working at the moment. Thanks for your patience.

hwiedPro, 2023-01-20:
This is solved for v2.0.0. Therefore I close this issue now.

commit be8328bb (2023-01-23): Fix #225
https://github.com/dfki-ric/phobos/issues/225
closed 2023-01-20 by hwiedPro