dfki-ric/phobos #226: Error when loading urdf file for robot hands

YoruCathy, 2021-06-22:
**Describe the bug**
When loading URDF, the mesh parts are loaded but in a messy order, and the code returns errors and warnings.

**To Reproduce**
1. Install and activate phobos add-on
2. run`bpy.ops.phobos.import_robot_model(filepath="D:/robot/hand_models/dh3_urdf/urdf/dh3_urdf.urdf")`in scriptting

**Screenshots**
The robot hand becomes like this after urdf is loaded, which is not the right structure.
[image]

[image]

**Desktop:**
- OS: Win10
- Blender version 2.93
- I use Phobos in this branch: https://github.com/dfki-ric/phobos/tree/blender2.9

**Additional context**
Errorlog:
[20210622_22:42:20] [1m[91mERROR[0m Can not create inertial object.[2m (C:\Users\cathy\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\phobos\model\inertia.py - createInertial (l49))[0m
[20210622_22:42:20] [1m[91mERROR[0m This path D:\robot\hand_correspondance\hand_models\dh3_urdf\urdf\package:\dh3_urdf\meshes\base_Link.STL is no file. Object visual_0_base_Link will have empty mesh![2m (C:\Users\cathy\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\phobos\model\geometries.py - createGeometry (l153))[0m
[20210622_22:42:20] [1m[91mERROR[0m Material '' is not defined.[2m (C:\Users\cathy\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\phobos\model\materials.py - assignMaterial (l115))[0m
[20210622_22:42:20] [1m[91mERROR[0m This path D:\robot\hand_correspondance\hand_models\dh3_urdf\urdf\package:\dh3_urdf\meshes\base_Link.STL is no file. Object collision_0_base_Link will have empty mesh![2m (C:\Users\cathy\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\phobos\model\geometries.py - createGeometry (l153))[0m
[20210622_22:42:20] [1m[91mERROR[0m Can not create inertial object.[2m (C:\Users\cathy\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\phobos\model\inertia.py - createInertial (l49))[0m
[20210622_22:42:20] [1m[91mERROR[0m This path D:\robot\hand_correspondance\hand_models\dh3_urdf\urdf\package:\dh3_urdf\meshes\finger1_base_Link.STL is no file. Object visual_0_finger1_base_Link will have empty mesh![2m (C:\Users\cathy\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\phobos\model\geometries.py - createGeometry (l153))[0m
[20210622_22:42:20] [1m[91mERROR[0m Material '' is not defined.[2m (C:\Users\cathy\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\phobos\model\materials.py - assignMaterial (l115))[0m
[20210622_22:42:20] [1m[91mERROR[0m This path D:\robot\hand_correspondance\hand_models\dh3_urdf\urdf\package:\dh3_urdf\meshes\finger1_base_Link.STL is no file. Object collision_0_finger1_base_Link will have empty mesh![2m (C:\Users\cathy\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\phobos\model\geometries.py - createGeometry (l153))[0m
[20210622_22:42:20] [1m[91mERROR[0m Can not create inertial object.[2m (C:\Users\cathy\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\phobos\model\inertia.py - createInertial (l49))[0m
[20210622_22:42:20] [1m[91mERROR[0m This path D:\robot\hand_correspondance\hand_models\dh3_urdf\urdf\package:\dh3_urdf\meshes\finger1_Link.STL is no file. Object visual_0_finger1_Link will have empty mesh![2m (C:\Users\cathy\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\phobos\model\geometries.py - createGeometry (l153))[0m
[20210622_22:42:20] [1m[91mERROR[0m Material '' is not defined.[2m (C:\Users\cathy\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\phobos\model\materials.py - assignMaterial (l115))[0m
[20210622_22:42:20] [1m[91mERROR[0m This path D:\robot\hand_correspondance\hand_models\dh3_urdf\urdf\package:\dh3_urdf\meshes\finger1_Link.STL is no file. Object collision_0_finger1_Link will have empty mesh![2m (C:\Users\cathy\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\phobos\model\geometries.py - createGeometry (l153))[0m
[20210622_22:42:20] [1m[91mERROR[0m Can not create inertial object.[2m (C:\Users\cathy\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\phobos\model\inertia.py - createInertial (l49))[0m
[20210622_22:42:20] [1m[91mERROR[0m This path D:\robot\hand_correspondance\hand_models\dh3_urdf\urdf\package:\dh3_urdf\meshes\finger1_tip_Link.STL is no file. Object visual_0_finger1_tip_Link will have empty mesh![2m (C:\Users\cathy\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\phobos\model\geometries.py - createGeometry (l153))[0m
[20210622_22:42:20] [1m[91mERROR[0m Material '' is not defined.[2m (C:\Users\cathy\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\phobos\model\materials.py - assignMaterial (l115))[0m
[20210622_22:42:20] [1m[91mERROR[0m This path D:\robot\hand_correspondance\hand_models\dh3_urdf\urdf\package:\dh3_urdf\meshes\finger1_tip_Link.STL is no file. Object collision_0_finger1_tip_Link will have empty mesh![2m (C:\Users\cathy\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\phobos\model\geometries.py - createGeometry (l153))[0m

Error: Python: Traceback (most recent call last):
  File "C:\Users\cathy\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\phobos\operators\io.py", line 285, in execute
    models.buildModelFromDictionary(model)
  File "C:\Users\cathy\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\phobos\model\models.py", line 1306, in buildModelFromDictionary
    bpy.ops.view3d.view_selected()
  File "D:\blender\2.93\scripts\modules\bpy\ops.py", line 132, in __call__
    ret = _op_call(self.idname_py(), None, kw)
RuntimeError: Operator bpy.ops.view3d.view_selected.poll() expected a view3d region

location: D:\blender\2.93\scripts\modules\bpy\ops.py:132

Traceback (most recent call last):
  File "<blender_console>", line 1, in <module>
  File "D:\blender\2.93\scripts\modules\bpy\ops.py", line 132, in __call__
    ret = _op_call(self.idname_py(), None, kw)
RuntimeError: Error: Python: Traceback (most recent call last):
  File "C:\Users\cathy\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\phobos\operators\io.py", line 285, in execute
    models.buildModelFromDictionary(model)
  File "C:\Users\cathy\AppData\Roaming\Blender Foundation\Blender\2.93\scripts\addons\phobos\model\models.py", line 1306, in buildModelFromDictionary
    bpy.ops.view3d.view_selected()
  File "D:\blender\2.93\scripts\modules\bpy\ops.py", line 132, in __call__
    ret = _op_call(self.idname_py(), None, kw)
RuntimeError: Operator bpy.ops.view3d.view_selected.poll() expected a view3d region

location: D:\blender\2.93\scripts\modules\bpy\ops.py:132

malter, 2021-06-22:
Hey, I just checked the URDF import on Windows10 with Blender 2.93 with a rather complex model and it was working fine. Could you share your model or another small one that is not working on your system? Once I can reproduce the problem I might be able to fix it. ;-)

YoruCathy, 2021-06-22:
Great! I put the urdf in google drive. 
https://drive.google.com/drive/folders/1pjpMU0irj_YzmC95j279X1DfiNXLYFjd?usp=sharing

malter, 2021-06-23:
Thank you for the model. Unfortunately, Phobos can not deal with `package://` file path from ROS. So if you replace "package://dh3_urdf/" with the relative path "../" in the URDF file the model should load fine. I don't yet have a clue how to deal with the url correctly. Something like searching the package root by searching for the `package.xml`?

YoruCathy, 2021-06-23:
Oh, I see, that is a problem. I tried to replace the 'package://' with the path the mesh files are, but that will also cause some errors. 
For example, the shadowhand model, I replaced the path, and loaded it. 
[image]

[image]
I think the inertial should not affect visualization, but the mesh is still messy.
[image]
Can you load the shadowhand model correctly?

malter, 2021-06-23:
Ok, here you found the next issue ;-)
Phobos is not importing the mesh scale correctly, maybe due to the adaptations to the new Blender version. So I had to scale down all meshes by a factor of `0.1`. Also the DAE materials are not loaded correctly, after removing the `black_spec.00x` materials from the meshes in Blender it looks not that bad:
[image]
[image]

YoruCathy, 2021-06-23:
WOW yours looks awesome, but I still cannot do it. Are you adjusting the scale here? 
[image]

malter, 2021-06-23:
Yes, but sorry the factor was `0.01` so changing `0.1` to `0.001`.

YoruCathy, 2021-06-24:
It works! Thanks a lot.

closed 2021-06-24 by YoruCathy