dfki-ric/phobos #228: Joint Creation Failed

tahir1069, 2021-06-24:
I am trying to create a URDF of tracked vehicle in Blender with the help of Phobos.  At the moment I am following [this}(https://www.youtube.com/watch?v=CrMvtlZl3LI&t=205s) tutorial. When creating joints for the robot(also if I set the name) I get an error *root object not found*. The detailed error is as follows:

```bash
/home/mbs/blender-2.79b/phobos/mbs_models/resources/resources.blend
Traceback (most recent call last):
  File "/home/mbs/.config/blender/2.79/scripts/addons/phobos/operators/naming.py", line 120, in execute
    root.pose.bones[0].custom_shape = ioUtils.getResource(('link', 'root'))
  File "/home/mbs/.config/blender/2.79/scripts/addons/phobos/utils/io.py", line 416, in getResource
    newobjects = importResources((specifiers,))
  File "/home/mbs/.config/blender/2.79/scripts/addons/phobos/utils/io.py", line 386, in importResources
    with bpy.data.libraries.load(filepath) as (data_from, data_to):
OSError: load: /home/mbs/blender-2.79b/phobos/mbs_models/resources/resources.blend failed to open blend file

location: <unknown location>:-1

location: <unknown location>:-1
[20210624_12:36:13] ERROR No root object found! Check your object selection. (phobos/utils/selection.py - getRoot (l210))
[20210624_12:36:13] ERROR No root object found! Check your object selection. (phobos/utils/selection.py - getRoot (l210))

```
I have also checked on the wiki of Phobos but still no success as the video is the same as in wiki. 

Any suggestions on the issue will highly be appreciated.
 
[image]

malter, 2021-06-24:
Does the documentation in the wiki help: https://github.com/dfki-ric/phobos/wiki/Kinematic-Skeleton?
Last time I tested it, it was working smoothly. So, try to setup the armature structure, do the parenting (bone relative), select the Phobos type for the armatures and try to set the robot name again while only the armatures are selected.

malter, 2021-12-15:
Closed due to no further feedback, if the problem still exists the issue can be reopened.

closed 2021-12-15 by malter