dfki-ric/phobos #237: Where is the export panel?

ivan-kapelyukh, 2022-04-06:
Hello! I am using Phobos to export a model to URDF. However, the export panel is not showing anywhere for me. Perhaps this is because I am using Blender 2.93 (with the current master version of Phobos), and the Wiki instructions for exporting are for a different version of Blender (maybe 2.7). I have installed and activated Phobos following the instructions, and can see Phobos object details like "Phobos Matrix Information", but no export panel. There are also no Phobos tabs at the top.

Can you please help me out with finding the export panel? Many thanks!

[image]

malter, 2022-04-11:
Hey, you can find a very small arrow on the right side of the colored coordinate system in the 3d-view. If you click on that arrow a tool panel will show up where you can also find a Phobos tab.

ivan-kapelyukh, 2022-04-11:
That works, thanks very much!

closed 2022-04-11 by ivan-kapelyukh