dfki-ric/phobos #238: Blender is now on 3.1.x

vkuehn, 2022-04-18:
Blender is now on Version above 3.x the project for 2.8 migration is still open.
I really would appreciate the functionality in the latest blender versions !

Is that project still active ?

kavonszadkowski, 2022-04-19:
It's an ongoing effort as part of a further development of phobos with the master branch working towards that goal. We still need to discuss how to deal with Blender's rapid release cycle these days.

malter, 2022-04-19:
As Kai mentioned you can try out the master branch, it should be feature complete for most general applications and is working with Blender 2.8 and above (also 3.1). At least, I just tested to load a blender file (model) that was created with Blender 2.8 and I was able to export the URDF with Phobos in Blender 3.1. Also the URDF import seems to work. However, I haven't tested all single operators but I won't expect much issues there, too. :-)

So the project is still active. However, the pace is currently not as desired but we expect some more resources this year. ;-)

vkuehn, 2022-04-19:
great news and  thanks for the project anyways !

closed 2022-04-19 by vkuehn