dfki-ric/phobos #240: Blender 3.1 does not show phobos after import

vkuehn, 2022-04-30:
**Describe the bug**
Blender does not find phobos add on
**To Reproduce**
Steps to reproduce the behavior:
1.https://github.com/dfki-ric/phobos.git
2.zip -r phobos.zip
3.in Blender 3.1 Edit -< Preferences -> Add-ons -> choose phobos.zip
4.Serach phobos

**Expected behavior**
phobos is shown an can be activated

**Desktop (please complete the following information):**
 - OS: Ubuntu 20.04

malter, 2022-05-02:
Hey,

did you tried:
1. git clone https://github.com/dfki-ric/phobos.git
2. cd phobos
3. zip -r phobos.zip phobos

And then load the zip via Blender? Possibly, you missed the second step and zipped the cloned folder instead of the phobos subfolder?

vkuehn, 2022-05-02:
doh... that could be the case, following your steps did actually work.

Thanks and sorry

closed 2022-05-02 by vkuehn