dfki-ric/phobos #241: No Video Tutorials Page sind 2019

vkuehn, 2022-05-02:
The Video Tutorials Section is Empty since 2019. 

With the evolution of ros some new tutorials a greatly appreciated by newcomers like me.

Probably like the one here https://www.youtube.com/watch?v=JGPyNxzVlYA !?

Amudtogal, 2022-05-03:
Yes, I agree. I had prepared a whole video series for the [Capricious Choutengan release](https://github.com/dfki-ric/phobos/releases/tag/1.0.2), which must still be somewhere in the DFKI archives.
Sadly, it was never released on Youtube. @kavonszadkowski , could you check if it is still there?

vkuehn, 2022-07-04:
@Amudtogal @kavonszadkowski that would be greatly appreciated. Anythink I could do to add some progress ?