dfki-ric/phobos #253: Importing a URDF produces unconnected elements

akhilsathuluri, 2022-08-28:
I am trying to import the URDF of a leg of the robot [solo from ORDI](https://github.com/open-dynamic-robot-initiative/trifinger_simulation). However, the imported URDF is not connected as shown below,

[image]

Moreover, the rig isnt aligned to the links as expected from the robots kinematics. Are there some post processing steps that I am missing that needs to be done after the import?
I am using Blender 3.1.2 on Windows.

malter, 2022-08-29:
Hey, looks like there is a problem with the mesh import. Did you tried to remove the ROS path definition in the URDF with relative paths (replace `package://robot_properties_fingers/` with `../` in the URDF file)?

akhilsathuluri, 2022-08-30:
Yes, the URDF only contains the absolute path for the meshes, 
```
<mesh filename="../meshes/solo_upper_leg_right_side.obj"
```

Is there something else thats happening?

malter, 2022-09-01:
After replacing the pathes the urdf (https://github.com/open-dynamic-robot-initiative/trifinger_simulation/blob/master/trifinger_simulation/robot_properties_fingers/urdf/finger.urdf) was loading fine on my systems (MacOS and Windows). Where exactly can I find the urdf you are trying to load? Do you use the Phobos release: https://github.com/dfki-ric/phobos/releases/tag/1.0.3 ?

akhilsathuluri, 2022-09-03:
The issue is fixed now. There were some elements which still had the ``package://`` tag in the mesh path. Replacing all of them with a relative path as you mentioned fixed the issue. Thanks for your quick replies!

closed 2022-09-03 by akhilsathuluri