dfki-ric/phobos #259: URDF to fbx

tomkimsour, 2022-10-18:
**Is your feature request related to a problem? Please describe.**
I want to go from my urdf to a fbx file in order to use the robot in a game engine. Has anyone ever been through this process ? If so how. Because exporting the model we get into blender to fbx seems to mess up everything.

If it was possible to have a way to export properly to fbx that would be convinient.

hwiedPro, 2023-03-13:
Hi @tomkimsour,
we probably won't be able to support this feature directly from our side.
But it should be possible to write a conversion script, that goes through all the bones and orients them as necessary for fbx. In phobos they are oriented regarding the frames and joint constraints. But fbx needs them oriented like real bones, I guess.

Hence, I hearby open the discussion and we are waiting for contributions on adding such a script.

tomkimsour, 2023-05-10:
> Hi @tomkimsour 
> Why don't you want to go this way:
> - import URDF by phobos
> - export model as FBX by standart blender exporter
> ?

Hi,
I'm not working on this anymore but the idea was to import the model into Unreal5. And fbx was the format i needed.

tomkimsour, 2023-05-10:
Sorry I think I missread that. And that's actually what I did but the result was looking nothing like what it should have been looking like.

brothermechanic, 2023-05-11:
> Sorry I think I missread that. And that's actually what I did but the result was looking nothing like what it should have been looking like.

By the pipeline, which I described above, you can get the some optimal result.

Looks like you expected the result as in AAA games?
For this you need manual shading/baking pipeline.
This has not yet been implemented, I just have it in my task list ;)

tomkimsour, 2023-05-11:
[image]
So this is what I was ending up with. The bones were completely up. On the right is the normal import of the urdf and on the left is the import of the fbx file generated by the export of the one on the left.