dfki-ric/phobos #265: Error on exporting URDF

jeffreyjene, 2022-12-12:
Error on export. 

[image]

 - OS: Windows 11
 - Version Blender 3.21

jeffreyjene, 2022-12-12:
Some more log info:

Traceback (most recent call last):
  File "C:\Users\jeffr\AppData\Roaming\Blender Foundation\Blender\2.90\scripts\addons\phobos\phobosgui.py", line 907, in draw
    proplist = dictprops.keys()
AttributeError: 'tuple' object has no attribute 'keys'

location: <unknown location>:-1

clwmrndl92, 2023-01-26:
I had the same error while exporting to urdf.  And, I did 'define geometry' for all visual objects, and they were exported normally.

hwiedPro, 2023-01-27:
@clwmrndl92 Thanks for your comment, that's true. We'll enhance the error message regarding this.
I close this as solved.

closed 2023-01-27 by hwiedPro

jeffreyjene, 2023-01-27:
Yes, thank you I figured out the issue.