dfki-ric/phobos #266: Cannot export URDF file

gy2256, 2023-01-29:
**Describe the bug**
When I click export button after all the setting nothing happened. 

**To Reproduce**
Steps to reproduce the behavior:
1. Click model
2. Hit "n"
3. Hit Export
4. Nothing happens. 

**Expected behavior**
URDF file will be produced. 

**Desktop (please complete the following information):**
 - OS: Mac OS Ventura (M1 Pro MacBook Pro)
 - Version 13.2
 - Blender: 3.4.0 (Apple Silicon Native)
 - Phobos 1.0.3

hwiedPro, 2023-01-31:
Have you selected in the viewport, what you want to export?

gy2256, 2023-01-31:
Yes. I pressed the 'a' and the model is selected. Nothing happened when I clicked 'a'.

hwiedPro, 2023-02-01:
Is everything highlighted in the viewport after pressing 'a'?

gy2256, 2023-02-01:
[image]
Thanks for the reply. This is what it looks like after I press 'a'.

hwiedPro, 2023-02-01:
It seems you have neither created a link, nor set a modelname and version, nor parented this visual to that link. Therefore you haven't yet created any structure that could be exported to URDF.

BTW: I can't see in your screenshot whether you have defined the geometry but if it shall be a mesh, you are currently not exporting any meshes due to your export settings, therefore your URDF will be invalid in the end.

Please see https://github.com/dfki-ric/phobos/wiki/Modeling-Walkthrough on how to create robot models.

hwiedPro, 2023-02-03:
I consider this as solved. Please feel free to reopen, if the issue persists.