dfki-ric/phobos #270: [Help] Phobos does not write joint to output URDF file

manas96, 2023-03-13:
I'm trying to create a simple model of a bike chain link with a single rotary joint. I followed the tutorial and I think I've created everything that is required (see the scene hierarchy):

[image]

However, when I export this to URDF, I get everything except the rotary joint. What could I be doing wrong?

Thanks!

hwiedPro, 2023-03-13:
It seems that your robot doesn't have a root link.
Joints can only be established between two links. As you have only link_bot this joint is lacking the parent link.

Or put in other words: Create ony link for the right chain-link one link for the left chain-link. Then parent the corresponding visuals and collisions to the respective links. and in the next step parent one link to the other. Then for the child link a joint can be defined.

We will add an error message for this faulty joint definition in v2.0.0.

closed 2023-03-13 by hwiedPro

manas96, 2023-03-13:
Thanks for the quick response! I really like this tool, and I think you are helping out a great number of people :)
You're right - I have no parent-child relationship for defining a joint. I followed your suggestions and got this hierarchy which to me looks correct (but is probably wrong):
[image]

One of the links also changed its visual shape which I believe indicates it is the root link.

However, this produces the following XML:
<?xml version="1.0"?>

  <robot name="should_work">

    <link name="new_link">
      <inertial>
        <origin xyz="0.32338 1.14848 0.00245" rpy="0 0 0"/>
        <mass value="3.121"/>
        <inertia ixx="0.001" ixy="0" ixz="0" iyy="0.001" iyz="0" izz="0.001"/>
      </inertial>
      <visual name="bot">
        <origin xyz="0.14638 0.33418 0.00111" rpy="1.5708 -1.57838 0"/>
        <geometry>
          <mesh filename="..\meshes\dae\Object38.dae" scale="1.0 1.0 1.0"/>
        </geometry>
      </visual>
      <collision name="bot">
        <origin xyz="0.14394 0.32918 0.00109" rpy="1.5708 -1.57838 0"/>
        <geometry>
          <mesh filename="..\meshes\dae\Object38.001.dae" scale="1.0 1.0 1.0"/>
        </geometry>
      </collision>
    </link>

    <joint name="new_link.000" type="revolute">
      <origin xyz="-0.00369 0.01257 -1e-05" rpy="0 -3.137 0"/>
      <parent link="new_link"/>
      <child link="new_link"/>
      <axis xyz="0 0 1.0"/>
      <limit lower="-3.14" upper="3.14" effort="0" velocity="0"/>
    </joint>

  </robot>
It seems the second link (`new_link.000`) is not being written to the file. I checked the validate method in Phobos, and it returned true. And there are no Phobos errors in the log. What am I missing?

hwiedPro, 2023-03-14:
The way you defined your links looks a bit weird as they are in the same place, but hey, I don't know your purpose.
Please check in the custom properties of new_link.000 if there is a link/name entry and if so delete it. Then try again.

manas96, 2023-03-14:
I have another question: How do I create a joint that's "fixed" to the world? I.e. imagine a gear that is attached to the world in its center, and it has a revolute joint there:
[image]

I read the wiki but couldn't find anything regarding this. I assume I'll need to create another link that's "fixed", but I don't know how to specify that in Phobos.

hwiedPro, 2023-03-15:
The fixture to the world happens in the setup/configuration of the simulation itself. How this works depends on the simulator you are using.

Please note that we can only give support regarding phobos itself. For the usage of multibody simulations refer to books, lectures and/or youtube tutorials on this topic. The manuals of the respective simulators should also provide help on how to tackle your problem.

manas96, 2023-03-15:
Understood, thank you!