dfki-ric/phobos #273: Relative path of meshes in SDF export not working

dettmann, 2023-03-31:
**Describe the bug**
When you export a sdf model you cannot export relative mesh paths. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Phobos's Export'
2. Check 'sdf' and 'dae' as export options and select 'relative' file path for meshes.
3. Clkick 'Export Model'
4. Hava a look at the exported sdf file.

**Expected behavior**
In visual section of a link, I would expect some relative link to the mesh like:

    <geometry>
        <mesh>
              <uri>../meshes/dae/<mesh_name>.dae</uri>
              <scale>1.0 1.0 1.0</scale>
         </mesh>
    </geometry>

. Instead a model path is referenced like:

    <geometry>
        <mesh>
              <uri>model://<model_name>/meshes/<mesh_name>.dae</uri>
              <scale>1.0 1.0 1.0</scale>
         </mesh>
    </geometry>

, which I would expect when you select "ros_package" as file path option.

**Desktop**
 - OS: Ubuntu 22.04
 - Blender 3.01
 - Phobos git has id: 8731dcfcc323571b654b25f02bc081efb7578174

hwiedPro, 2023-04-03:
Hi Alex,
please try the branch pre_v2.0.0. I'd appreciate every tester there ;)

dettmann, 2023-04-04:
Hi Henning,

thanks for the quick reply. I just tested and had following issues:

1. the update process was maybe not clear to me, but installing the new phobos.zip add-on in the blender preferences lead to the fact that the phobos gui elements were vanished. I could only solve it by re-installing blender again.

2. After a fresh blender install and installing the phobos add-on, during activation the following error occured:

    File "/home/dettmann/.config/blender/3.0/scripts/addons/phobos/blender/phobosgui.py", line 77, in PhobosPrefs default=os.getlogin(),
    FileNotFoundError: [Errno 2] No such file or directory
    
    Maybe this is related to my WSL usage? However, I solved this by manually setting my user name, for now.

3. Then, the installation worked, but following error was printed (and ignored by me):

    Python: Traceback (most recent call last):
      File "/home/dettmann/.config/blender/3.0/scripts/addons/phobos/blender/phobosgui.py", line 789, in draw
        errors = validation.validateObjectNames(obj)
    AttributeError: module 'phobos.blender.utils.validation' has no attribute 'validateObjectNames'. Did you mean: 'validateObjectPose'?
    
4. The export of an existing model was working after redefining the link as motor. But the export misses the xml and sdf version, the first only being important for syntax highlighting whereas the second missing prevents gazebo from parsing the file. Adding manually something like this works:

<?xml version="1.0"?>
<sdf version="1.5">

5. The export option "Force almost zero to zero" does not work.

6. The loaded model is black in gazebo, i.e. the mesh colors are not transorted. However, this might be another issue since it was completely grey before (which was also not correct).

I hope my test report helps you to improve this wonderful software ;)

dettmann, 2023-04-05:
Hi Henning,

thanks for the quick fixes :) The update process as described above worked as well. 

Best, Alex

closed 2023-04-11 by hwiedPro