dfki-ric/phobos #278: [pre_v2.0.0] Create Collision Objects by mesh

brothermechanic, 2023-05-03:
**Describe the bug**
Create Collision Objects by mesh don't works (any other options like cube - works)

**To Reproduce**
Steps to reproduce the behavior:
1. Run bpy.ops.phobos.create_collision_objects() ("Create Collision Objects")
2. Select type - Mesh
4. See error
```python
Error: Python: Traceback (most recent call last):
  File "/home/bm/.config/blender/3.5/scripts/addons/phobos/blender/operators/editing.py", line 1343, in execute
    bpy.ops.object.transform_apply(location=False, rotation=False, scale=True, properties=False)
  File "/usr/share/blender/3.5/scripts/modules/bpy/ops.py", line 113, in __call__
    ret = _op_call(self.idname_py(), None, kw)
RuntimeError: Error: Cannot apply to a multi user: Object "grip_rotator_collision", Mesh "Part__Feature007.003", aborting
```

**Expected behavior**
In Phobos v1 it works

**Desktop:**
 - OS: Gentoo
 - Version: ~amd64

brothermechanic, 2023-05-04:
@hwiedPro  Hi,
Error message has changed
```python
Python: Traceback (most recent call last):
  File "/home/bm/.config/blender/3.5/scripts/addons/phobos/blender/operators/editing.py", line 1282, in execute
    phobos_vis = blender2phobos.deriveVisual(vis)
  File "/home/bm/.config/blender/3.5/scripts/addons/phobos/blender/io/blender2phobos.py", line 189, in deriveVisual
    material = deriveMaterial(obj.active_material, logging=logging)
  File "/home/bm/.config/blender/3.5/scripts/addons/phobos/blender/io/blender2phobos.py", line 50, in deriveMaterial
    if "No material defined." in errors:
TypeError: argument of type 'NoneType' is not iterable
```

brothermechanic, 2023-05-10:
@hwiedPro  new error message

```python
Error: Python: Traceback (most recent call last):
  File "/home/bm/.config/blender/3.5/scripts/addons/phobos/blender/operators/editing.py", line 1318, in execute
    origin=representation.Pose.from_matrix(phobos_vis.origin.to_matrix().dot(transform))
TypeError: Pose.from_matrix() missing 1 required positional argument: 'relative_to'
```

i think collision wants be relative from armature object, but can't find it

hwiedPro, 2023-05-11:
Perfect that you have found that. To facilitate more profound SDF support I recently made the relative_to property mandatory for all Poses. I thought I had found all usages but some seem to have slipped through. I'll push the fix asap

brothermechanic, 2023-05-11:
```ptyhon
Python: Traceback (most recent call last):
  File "/home/bm/.config/blender/3.5/scripts/addons/phobos/blender/operators/editing.py", line 1318, in execute
    origin=representation.Pose.from_matrix(phobos_vis.origin.to_matrix().dot(transform))
TypeError: Pose.from_matrix() missing 1 required positional argument: 'relative_to'
```

brothermechanic, 2023-05-11:
Maybe I'm doing something wrong ?
https://github.com/dfki-ric/phobos/assets/3109995/d8405b2c-8b01-4fbd-a736-287859300d21
[test_model.zip](https://github.com/dfki-ric/phobos/files/11450049/test_model.zip)

hwiedPro, 2023-05-11:
Thanks, this was a not yet handled blender color type in the color parser. Should be fixed now.

hwiedPro, 2023-05-30:
I consider this as solved, please reopen if not.

closed 2023-05-30 by hwiedPro