dfki-ric/phobos #281: [pre_v2.0.0] Create inertial objects

brothermechanic, 2023-05-10:
**Describe the bug**
Create inertial objects don't work

```python
Error: Python: Traceback (most recent call last):
  File "/home/bm/.config/blender/3.5/scripts/addons/phobos/blender/operators/editing.py", line 1318, in execute
    origin=representation.Pose.from_matrix(phobos_vis.origin.to_matrix().dot(transform))
TypeError: Pose.from_matrix() missing 1 required positional argument: 'relative_to'
```
**To Reproduce**
Steps to reproduce the behavior:
1. Select object for inertia calculation
2. Click on "Create inertials" (bpy.ops.phobos.generate_inertial_objects())
3. See error

```python
Error: Python: Traceback (most recent call last):
  File "/home/bm/.config/blender/3.5/scripts/addons/phobos/blender/operators/editing.py", line 1318, in execute
    origin=representation.Pose.from_matrix(phobos_vis.origin.to_matrix().dot(transform))
TypeError: Pose.from_matrix() missing 1 required positional argument: 'relative_to'
```

**Expected behavior**
In Phobos v1 it works

**Desktop (please complete the following information):**
 - OS: Gentoo
 - Version: ~amd64

hwiedPro, 2023-05-11:
Yep as already written on the other issue. I just made relative_to mandatory to facilitate better SDF support, some of the calls slipped through during refactoring will be fixed asap.

hwiedPro, 2023-05-11:
Ok, please try again ;)

brothermechanic, 2023-05-11:
```python
Python: Traceback (most recent call last):
  File "/home/bm/.config/blender/3.5/scripts/addons/phobos/blender/operators/editing.py", line 1194, in execute
    inertial = representation.Inertial(
  File "/home/bm/.config/blender/3.5/scripts/addons/phobos/io/representation.py", line 1406, in __init__
    assert self.origin.relative_to is not None
AssertionError

```

brothermechanic, 2023-05-11:
@hwiedPro 
Maybe I'm doing it wrong
I would also like that in case of a user mistake, there will be tooltip instead of debugging information python

hwiedPro, 2023-05-11:
Nope, you are doing it totally correct. I can't say how thankful I am for your testing and debugging. This speeds up my work a looot.
I (hopefully ;)) got the missing relative_tos for this case now fixed as well. Please, try again.

And yes you are right, we should in general have more tooltips and easier understandable error messages.

hwiedPro, 2023-05-30:
I consider this as solved, please reopen if not.

closed 2023-05-30 by hwiedPro