dfki-ric/phobos #286: Wrong behaviour when mirroring object

fra-cle, 2023-05-22:
**Describe the bug**
I have a skeleton (a right robot hand) that I would like to mirror to generate the left handed version of the model. However, even if the model is mirrored correctly ('CTRL+M'), the exported URDF is not working properly (the joints are misplaced).

**To Reproduce**
Steps to reproduce the behavior:
    Activate object mode
    Select model
    Select hierarchy
    Mirror (CTRL+M + X)

The model is mirrored but does not either show or work correctly in RVIZ

**Expected behavior**
The URDF model should have shown correctly in RVIZ

**Desktop:**
 - OS: Windows 
 - Version: 10
 - Blender version: 2.90.1
 - Phobos version: 1.0.1

hwiedPro, 2023-06-15:
With version phobos 2.0.0 you can do mirroring of complete robot structures like this:
```python
import phobos
r = phobos.core.Robot(inputfile="your/robot/file.urdf")
r.mirror_model(mirror_plane=[0,1,0])
r.export_xml("your/robot/mirrored", format="urdf", with_meshes=True)
```
this should give you an `urdf/` and `meshes/` directory in `your/robot/mirrored`.

I consider this issue as solved, if not please reopen. ;)

closed 2023-06-15 by hwiedPro

fra-cle, 2023-06-26:
Hello, sorry I did not see this message before. I tried to use pre_v2.0.0 and on blender 3.3LTS but the problem is that the model has to be adapted (I understood that the default axis of rotation of the joints changed from Z to X or vice-versa, and could not do that easily). Would you please indicate if your solution would work also with a model made with an older version of Phobos?

Additionally, can you briefly explain better how to use the commands you indicate? Do you mean to do that in a python CLI?

fra-cle, 2023-06-26:
> @fra-cle Thanks for reporting this issue. I'm not sure whether this feature was ever related supported. However, it would be nice if you could try it with pre_v2.0.0 and on blender 3.3LTS.
> 
> And if it doesn't work please add the resulting URDF or even better a minimal example.

I tried to test the same thing with pre_v2.0.0 and on blender 3.3LTS but, as mentioned in the previous comment, the model is not compatible and does not work properly without adaptation (i.e. it does not export to URDF). Is there a quick way to make it work? Thanks

hwiedPro, 2023-06-26:
Hi @fra-cle,
no worries ;)
In blender there is no such support yet.
If you store your model as URDF/SDF/SMURF etc. you can do the mirroring as stated in my previous answer no matter with which phobos version the models have been created.
And yes you can simply do pip install . in your phobos directory and then execute the above code in any python script.

If you are mirroring stuff in blender and you want to adapt the joint axes you can do this with the "Define Joint"-Operator.

fra-cle, 2023-06-29:
> With version phobos 2.0.0 you can do mirroring of complete robot structures like this:
> 
> ```python
> import phobos
> r = phobos.core.Robot(inputfile="your/robot/file.urdf")
> r.mirror_model(mirror_plane=[0,1,0])
> r.export_xml("your/robot/mirrored", format="urdf", with_meshes=True)
> ```
> 
> this should give you an `urdf/` and `meshes/` directory in `your/robot/mirrored` and there an urdf named like your robot.
> 
> I consider this issue as solved, if not please reopen. ;)

Thank you again, I confirm that we followed your suggestion and were able to mirror the model. We only had to manually adjust the rotation direction of each axis (adding a minus sign) and all was set.

hwiedPro, 2023-06-29:
Hi @fra-cle ,

I'm happy that you succeded.
Yes, as we need to ensure a right-hand coordinate frame, one frame-axis will be flipped after mirroring automatically. There are options to mirror_model, to solve this directly. But setting the axes afterward mirroring manually is often the more straight forward solution.

All the best