dfki-ric/phobos #287: Trying to export a model to gazebo

marioesteban11, 2023-05-28:
Hi everyone,

I'm new using this addone and I have some doubts.

I'm trying to emulate in GAZEBO and object thet I will use to support an exoskeleton. To do this I need an .urdf file for gazebo and I think this repo is the most usefull one.

Im using blender 3.5.1 in ubuntu 20.04.

I try to install this addone on my blender but this error shows up: fake module--addon missing,'bl-info'give bad performance!

And everytime i try to give a <visual> or <colission> label this error shows up too:

AttributeError: 'tuple' object has no attribute 'keys'
Traceback (most recent call last):
  File "/home/marioest/.config/blender/3.5/scripts/addons/phobos/blender/phobosgui.py", line 907, in draw
    proplist = dictprops.keys()
AttributeError: 'tuple' object has no attribute 'keys'

My model is quite simple, i just need to give it a hitbox.

[image]

Do you have any hint or domething to help me? :)

Thanks in advance

marioesteban11, 2023-05-31:
Hi @hwiedPro,

Can you explain me how to install this pre_v2.0.0 branch? 

Should i just clone the repo in my pc and add the addone in blender or i need to do anything else?

Thanks in advance

hwiedPro, 2023-05-31:
Sure,
as explained in the READMEs:
1. clone on pre_v2.0.0 branch
2. zip phobos subdirectory
3. install addon in blender
4. restart blender
5. activate addon

marioesteban11, 2023-05-31:
I try those steps and this error shows up when I try to do steo 3:

Read prefs: /home/marioest/.config/blender/3.5/config/userpref.blend
addon not loaded: 'phobos'
cause: 
Read blend: /home/marioest/Desktop/TFG/cosas_3d/exo_palo.blend
fake_module: addon missing 'bl_info' gives bad performance!: '/home/marioest/.config/blender/3.5/scripts/addons/setup.py'
fake_module: addon missing 'bl_info' gives bad performance!: '/home/marioest/.config/blender/3.5/scripts/addons/testrunner.py'
fake_module: addon missing 'bl_info' gives bad performance!: '/home/marioest/.config/blender/3.5/scripts/addons/setup.py'
fake_module: addon missing 'bl_info' gives bad performance!: '/home/marioest/.config/blender/3.5/scripts/addons/testrunner.py'
fake_module: addon missing 'bl_info' gives bad performance!: '/home/marioest/.config/blender/3.5/scripts/addons/setup.py'
fake_module: addon missing 'bl_info' gives bad performance!: '/home/marioest/.config/blender/3.5/scripts/addons/testrunner.py'
Modules Installed () from '/home/marioest/phobos.zip' into '/home/marioest/.config/blender/3.5/scripts/addons'

marioesteban11, 2023-05-31:
When it seems to be installed, when i run blender this shows up:

blender
Read prefs: /home/marioest/.config/blender/3.5/config/userpref.blend
Read blend: /home/marioest/Desktop/TFG/cosas_3d/exo_palo.blend
fake_module: addon missing 'bl_info' gives bad performance!: '/home/marioest/.config/blender/3.5/scripts/addons/setup.py'
fake_module: addon missing 'bl_info' gives bad performance!: '/home/marioest/.config/blender/3.5/scripts/addons/testrunner.py'
/home/marioest/.config/blender/3.5/scripts/modules/numpy/core/getlimits.py:518: UserWarning: The value of the smallest subnormal for <class 'numpy.float64'> type is zero.
  setattr(self, word, getattr(machar, word).flat[0])
/home/marioest/.config/blender/3.5/scripts/modules/numpy/core/getlimits.py:89: UserWarning: The value of the smallest subnormal for <class 'numpy.float64'> type is zero.
  return self._float_to_str(self.smallest_subnormal)
/home/marioest/.config/blender/3.5/scripts/modules/numpy/core/getlimits.py:518: UserWarning: The value of the smallest subnormal for <class 'numpy.float32'> type is zero.
  setattr(self, word, getattr(machar, word).flat[0])
/home/marioest/.config/blender/3.5/scripts/modules/numpy/core/getlimits.py:89: UserWarning: The value of the smallest subnormal for <class 'numpy.float32'> type is zero.
  return self._float_to_str(self.smallest_subnormal)
Importing phobos
IMPORT:  phobos.blender.defs
Parsing definitions from: /home/marioest/.config/blender/3.5/scripts/addons/phobos/data/blender/definitions
  defaultJoints.yml
  defaultSubmechanisms.yml
  defaultMaterials.yml
  defaultSensors.yml
  defaultControllers.yml
  defaultMotors.yml
Creating new definition type: joints
Creating new definition type: motors
IMPORT:  phobos.blender.display
IMPORT:  phobos.blender.io
RELOAD:  phobos.blender.model
IMPORT:  phobos.blender.operators
RELOAD:  phobos.blender.phobosgui
RELOAD:  phobos.blender.phoboslog
RELOAD:  phobos.blender.phobossystem
RELOAD:  phobos.blender.reserved_keys
RELOAD:  phobos.blender.utils
Registering operators.selection... 
Registering operators.io...
Registering operators.editing...
TypeError: EnumProperty(...): expected a tuple containing (identifier, name, description) and optionally an icon name and unique number
ValueError: bpy_struct "PHOBOS_OT_add_motor" registration error: 'template' EnumProperty could not register (see previous error)
TypeError: EnumProperty(..., default='mechanism'): not found in enum members
ValueError: bpy_struct "PHOBOS_OT_define_submodel" registration error: 'submodeltype' EnumProperty could not register (see previous error)
Registering operators.naming...
Registering operators.misc...

Registering phobosgui...
  ... successful.

marioesteban11, 2023-05-31:
Do you have any tutorial for this new phobos version with this blender version? 

And how do I export an urdf after modifing my model.

Thanks in advance

hwiedPro, 2023-05-31:
In usage there are not many differences compared to the former phobos versions, so you can have a look at the README and the old wiki. However, in the wiki there is a pre_v2.0.0 branch as well.

pre_v2.0.0 will go to master in the next days.

For general usage questions please open a Q&A [here](https://github.com/dfki-ric/phobos/discussions/new?category=q-a)

As the technical issue is solved, I'll close this issue.

closed 2023-05-31 by hwiedPro