dfki-ric/phobos #291: Importing .urdf file not working as expected

Tindahood, 2023-06-09:
For a little side-project I wanted to see if I could get [this robot](https://github.com/unitreerobotics/unitree_ros/tree/master/robots/go1_description) into blender and see what's possible from there. I found this repository and saw that importing URDF is supported so I wanted to try. Upon importing though I get errors like these:
```
Python: Traceback (most recent call last):
  File "/home/me/.config/blender/2.93/scripts/addons/phobos/operators/io.py", line 283, in execute
    model = entity_io.entity_types[suffix]['import'](self.filepath)
  File "/home/me/.config/blender/2.93/scripts/addons/phobos/io/entities/urdf.py", line 620, in importUrdf
    log(" Adding link {}.".format(link.attrib['name']), 'DEBUG')
KeyError: 'name'

location: <unknown location>:-1
``` 
I feel like I'm this close to successfully importing but apparently some missing xml attributes seam to break the logging which in turn prevents a successful import.

Steps to reproduce the behavior:
1. From the [official LTS download page](https://www.blender.org/download/lts/2-93/) download [blender-2.93.18-linux-x64.tar.xz](https://www.blender.org/download/release/Blender2.93/blender-2.93.18-linux-x64.tar.xz) and this [phobos release](https://github.com/dfki-ric/phobos/releases/download/1.0.3/phobos.zip)
2. Clone the [unitree repo](https://github.com/unitreerobotics/unitree_ros) containing the robot files
3. extract blender, run it, install and activate the Phobos plugin just like any other plugin
4.  Open sidebar, select phobos, scroll down and click Import Robot Model
5. In the dialouge navigate to `/location/of/unitree_ros/robots/go1_description/urdf/` and select `go1.urdf`.
6. The error should appear in the bottom left corner

Using:
 - OS: Linux Mint
 - Version 21.1
 - Blender Version 2.93 LTS

hwiedPro, 2023-06-15:
Hi @Tindahood,
Can you do me a favour and try blender3.3.
Then clone this repo on branch pre_v2.0.0 and follow the readme for installation of phobos.
Thanks in advance ;)

Tindahood, 2023-06-15:
@hwiedPro I might find some time to do this soon. Doing this in a docker should work fine, right?

hwiedPro, 2023-06-15:
I have never tried using blender from  docker, but if you can connect to an xserver, it should work

hwiedPro, 2023-06-22:
As stated in https://github.com/dfki-ric/phobos/issues/296#issuecomment-1602637876, I tried to import the go1_description and it works for me with the phobos version from the master branch.

Therefore I close this. If you encounter further problems, feel free to specify and reopen.

closed 2023-06-22 by hwiedPro