dfki-ric/phobos #303: Do not install on Blender3.4

Gerka0291, 2023-08-07:
Do not instal after restart ((

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu]
 - Version [e.g. 16.04]
 - debugging output (run `setup.py --info`)

**Additional context**
Add any other context about the problem here.

E.g. URDF version, mesh type/version etc

hwiedPro, 2023-08-15:
Have you restarted Blender as stated in the instructions?

hwiedPro, 2023-09-01:
I consider this as solved, due to inactivity. Please reopen if further help is needed.

closed 2023-09-01 by hwiedPro

Developer310301, 2024-02-16:
I have the same issue mentioned before