dfki-ric/phobos #304: problem when installing dependency `lxml`: Python.h file not found

zhuoqun-chen, 2023-08-08:
## Describe the bug
When using the provided `install_requirements.py` to install the dependencies, one of the optional depency `lxml` failed to be installed because `Python.h` file is not found when building the wheel:

[image]

## To Reproduce
Steps to reproduce the behavior:
1. download `install_requirements.py` and go to the path where this script is downloaded
2. execute `$YOUR_BLENDER_EXECUTABLE -b —python install_requirements.py`
3. press `y` when prompted with the `Do you want to proceed? [y/n]` question
4. the bug occurs saying that `lxml` wheel failed to be built 

## Expected behavior
The optional dependency `lxml` can also be installed.

## Screenshots
See the description section above.

## Desktop (please complete the following information)
 - OS: `MacOS`
 - Version `Apple M1 Pro` Ventura 13.4`

## Additional context (Solution)
This issue can be viewed as a reference in case someone in the future has the same problem as me.
This problem can be fixed by following these steps:
1. find the python version your blender is using (`3.10.12` in my case)
2. use `conda` create an environment with the same python version as above, e.g., `conda create -n blender_ws python=3.10.12`) 
3. confirm that there is `Python.h` located inside `~/miniconda3/envs/blender_ws/include/python3.10/Python.h`
4. go into the terminal and add environment path `export C_INCLUDE_PATH=“~/miniconda3/envs/blender_ws/include/python3.10”`
5. execute the install script again within the same terminal: `$YOUR_BLENDER_EXECUTABLE -b —python install_requirements.py`
6. `lxml` should be successfully installed

hwiedPro, 2023-08-15:
Thanks  for reporting this and contributing a possible solution.

It seems blender on Mac is not properly installing python-dev, so it seems required to install this manually. Please not this similar issue and solution here on Linux: https://blender.stackexchange.com/a/107381

Also lxml is an optional dependency. If it's not available for you you can comment it from the requirements then the plain xml of python should be used.

hwiedPro, 2023-08-17:
I consider this as solved

closed 2023-08-17 by hwiedPro