dfki-ric/phobos #317: Cannot import certain urdf models into Blender v3.3 by the Phobos 2.0 version

edward9503, 2023-09-10:
Dear team,

recently I downloaded the Phobos from the master branch (which I think is the beta version of 2.0) and added it as an addon into blender v 3.3. I tried to import this [urdf](https://github.com/ANYbotics/anymal_c_simple_description/tree/master/urdf) into blender. But it always gives me error. I also tried with other urdf files. Some of them work but the others don't. I wonder if this is a bug for the beta version?

Many thanks.

hwiedPro, 2023-09-18:
Hi,
How is the urdf placed when you load it? Do you open it in it's package environment or have you only downloaded the urdf it self.

As your urdf contains package:// style pathes, phobos expects that the urdf is still in it's package. If you move it out of this package, it can not find the package and thus all other pathes are no longer valid.

edward9503, 2023-09-18:
Hi @hwiedPro ,
Thanks for the reply. I didn't use the package:// style anymore: I deleted all the package path, and only leave the mesh file name in the urdf. Then, I put the urdf together with the mesh files inside one single folder. So theoretically it is correct. I also tried with other robot model, it worked. It only cannot work with the Anymal C robot model. So I have no idea what is going on. I attached the model in anymalC.zip. Many thanks. 

[image]
[image]

[anymalC.zip](https://github.com/dfki-ric/phobos/files/12646553/anymalC.zip)

hwiedPro, 2023-09-18:
Hi @edward9503 ,
thanks for the reply.
It seems in the URDF there is a reference to material "orange" that is not defined in the URDF. That you should remove or specify.
However, there were some bugs in phobos as well. Thanks for pointing them out with your use case, they are fixed now. Please, fetch, zip and re-install phobos as stated in the README. 
For me it works now.
Looking forward to your reply.

edward9503, 2023-09-18:
Hi @hwiedPro ,

Thanks, it works now! 

Still, there is a little problem when importing this model. As you can see in the picture, the robot's base is penetrated by a linkage, which is not expected in urdf. Do you also see this issue from your side?

 
[image]

hwiedPro, 2023-09-18:
You are awesome! Thanks for testing.
I use DAE import quite rarely, so that was a good place for the bugs to hide. ;)
I just pushed a fix.

edward9503, 2023-09-18:
Hi @hwiedPro ,
Thanks for the update. Now the penetrating issue is solved. 
There is one last issue, which is when I just import the model, the hatch_visual part is very big (much more bigger than the robot model). As you can see the robot is overlapped in the middle.

[image]
[image]

hwiedPro, 2023-09-18:
Yes i noticed, I thought you put a room into it.  Sorry I'll check again

hwiedPro, 2023-09-18:
Yes, I know. I handle it now the same way as rviz, as I think for phobos dealing with the mesh like this is the more appropriate way. See https://github.com/dfki-ric/phobos/commit/bac32f3ff6edb745fbd3a1525dbc100035a7de8e ;)

edward9503, 2023-09-18:
@hwiedPro Thanks a lot! Now it looks perfect!!!

hwiedPro, 2023-09-18:
Thank you for testing. I'll close this then!

edward9503, 2023-09-19:
Hi @hwiedPro ,

I tried to import another robot model, but something with the scale problem occurs again. You can see the leg of the robot is giant. 
[image]

The model can be found here:
[aliengo.zip](https://github.com/dfki-ric/phobos/files/12655367/aliengo.zip)

Thanks.

hwiedPro, 2023-09-19:
Thanks, this seems now to be really an issue with the DAE.
In DAE it is possible to define the unit in which the vertices of the mesh are specified. Search for the tag unit.
It happens now that calf.dae uses this option. If you import it into blender directly (File->Import->Collada .dae) and select there Import units. You'll notice that this mesh is huge there as well. E.g. hip does not scale with the unit.
I guess rviz ignores this tag.

However there also seems to be a bug in blender itself. When loading the DAE without importing the units it's small as the scale is applied. Then when importing WITH units, it's big. But THEN, when importing again without units it's still big, the Import units option no longer takes effect.

edward9503, 2023-09-19:
Thanks. 

That's really an issue. Can we ignore that tag as well when using phobos to import the model?

hwiedPro, 2023-09-19:
Unfortunately phobos uses the exact same import units option. My suggestion here would be to export the calf.dae file again without the scale. I did so here: [calf.zip](https://github.com/dfki-ric/phobos/files/12656880/calf.zip)

I also opened an issue in blender here https://projects.blender.org/blender/blender/issues/112557

I hope there will be a solution soon.

edward9503, 2023-09-19:
Awesome, thanks for the help.

hwiedPro, 2023-09-21:
Hi @edward9503 ,
it should work now with the original aliengo model, too.
I'd be happy if you could confirm.

edward9503, 2023-09-22:
Hi @hwiedPro 
Yes it works on my side. Thanks so much.

closed 2023-09-22 by hwiedPro