dfki-ric/phobos #318: Wrong load in 1.0.3 in blender2.93

wungsui, 2023-09-20:
I use blende(2.93) and photos(1.0.3) in windows and I load a urdf of franka panda robot like this
[image]
but it came out to this:
[image]
it seems to the visual model and collision model are same,but in its urdf,it is:
[image]
the .dae and the .stl file is very different,then,I import all the .dae file and export to .stl file,and edit the urdf to this:
[image]
it seems to it has no color like this:
[image]

hwiedPro, 2023-09-21:
Have you also tried this with the current master (v2.0.0)

hwiedPro, 2024-02-13:
Closing this as this relates to an older no longer maintained version of phobos.

closed 2024-02-13 by hwiedPro