dfki-ric/phobos #322: Materials don't always get fully added to urdf

23pointsNorth, 2023-09-21:
**Describe the bug**
When exporting a model with a plain color material (principled BSDF, GGX, base color) per object, sometimes the last material doesn't get fully appended to the urdf - e.g. 

```
### All materials referenced correctly in URDF file

<material name="bucket_material">
  <color rgba="0.15686 0.06667 0.07451 1.00000"/>
</material>
<material name="swivel_material">
### end of file
``` 

**To Reproduce**
Steps to reproduce the behavior:
1. Create multiple objects with at least a couple of meshes in dae (unsure if relevant)
2. add material to them
3. export to URDF

**Expected behavior**
1. When exporting finishes, a message to the user to appear that it has completed (I assume on larger models it may take time and writing to the file buffer may not have flushed)
2. The full materials list to be completed.

**Desktop (please complete the following information):**
 - OS: Ubuntu Version [e.g. 22.04]
 - Blender version 3.3
 - Phobos version from 21 Sept

hwiedPro, 2023-09-25:
Hi @23pointsNorth ,
thanks for reporting your issue.
I tried reproducing this using the [aliengo](https://github.com/dfki-ric/phobos/files/12655367/aliengo.zip) model and edited the materials, added a new mesh, added a new material to that and exported everything to dae. However I couldn't reproduce your issue.

Could you please share more information? If possible your blend file or a minimal example. Also a screen shot of the material settings of swivel_material would help.

23pointsNorth, 2023-09-25:
The annoying part is that the issues is stochastic, where there may be times where it half flushes things, and sometimes pushes the whole file out.
 
I've attached the blend file. [machine.zip](https://github.com/dfki-ric/phobos/files/12712782/machine.zip)

And the material tab. 
[image]

P.S. The export paths in Linux (Ubuntu) by default are empty (best to set it to e.g. `./` or if relative is always prefixed by `//`, which is unnatural and e.g. if in `~`, then `//Downloads` is actually `/Downloads` and not `/home/{user}/Downloads`.

hwiedPro, 2023-09-25:
Ok I tried your example several times, for me it worked everytime.

Funny thought: Are you using gedit to view the urdf on Ubuntu? I sometimes have an issue with gedit, that it isn't able to scroll down to the very end of the file. Resizing the window or opening in another editor solves this.

23pointsNorth, 2023-09-25:
Generally I am using Sublime Text. Maybe it's something else on my end. 
Happy to close the issue unless I can consistently replicate.

23pointsNorth, 2023-09-25:
Unrelated, alas with the same blender file, adding an IMU sensor to the `base_link` is fine. Adding a GPS (to any link) results in the following error:

[image]

Am I missing something with the link tree and how sensors are added [Hardware -> Add Sensor -> name/category:GPS]?

hwiedPro, 2023-09-25:
As that what this issue was opened for seems solve I close this. (see https://github.com/dfki-ric/phobos/issues/322#issuecomment-1733472344)

closed 2023-09-25 by hwiedPro

23pointsNorth, 2023-09-26:
No worries, I was looking at the anybotics issue and assumed the preferred method is within a single issue.

> [FOV rads] Currently there is no option... So I guess you have to take the calculator...
Maybe I am within my own bubble, but where do folks use sdf/urdfs for? I've been in the ROS + Gazebo ecosystem and maybe those are some non-backwards compatible items that have occurred. (ROS Humble + Gazebo Garden 7.5)

hwiedPro, 2023-09-26:
> No worries, I was looking at the anybotics issue and assumed the preferred method is within a single issue.
I see, here it helps me getting my todo list straight and thereby  I can reference the issue in the fixing commit and others with the same problem can contribute to that issue. ;)

> Maybe I am within my own bubble, but where do folks use sdf/urdfs for? I've been in the ROS + Gazebo ecosystem and maybe those are some non-backwards compatible items that have occurred. (ROS Humble [...]

I'm no longer sure whether I understand your question.
Are you looking for a convenient method of typing angles in phobos or of specifing the type written to sdf/urdf?
I don't know for sdf but I'm pretty sure urdf can only handle radiant...

23pointsNorth, 2023-09-26:
> I'm no longer sure whether I understand your question. Are you looking for a convenient method of typing angles in phobos or of specifing the type written to sdf/urdf? I don't know for sdf but I'm pretty sure urdf can only handle radiant...

That was my intuition as well, that the sdf/urdf assumes radinats (but currently phobos exports degrees for the camera fov). 
So, I assumed there was another tool that uses the phobos output (not ros/gazebo), and thus it exporting the camera sensor in degrees being ok. So was curious if that was the case.