dfki-ric/phobos #324: Undefined key joint/type when exporting

literally-anything, 2023-09-25:
**Describe the bug**
Whenever I export, it gives me an error about the key joint/type not being defined:
```python
Python: Traceback (most recent call last):
  File "~/.config/blender/3.6/scripts/addons/phobos/blender/operators/io.py", line 141, in invoke
    return self.execute(context)
  File "~/.config/blender/3.6/scripts/addons/phobos/blender/operators/io.py", line 256, in execute
    self.exportModel(root, exportpath)
  File "~/.config/blender/3.6/scripts/addons/phobos/blender/operators/io.py", line 156, in exportModel
    robot = deriveRobot(root)
  File "~/.config/blender/3.6/scripts/addons/phobos/blender/io/blender2phobos.py", line 818, in deriveRobot
    joints=[deriveJoint(obj) for obj in objectlist if obj.phobostype == 'link' and sUtils.getEffectiveParent(obj, include_hidden=True) is not None],
  File "~/.config/blender/3.6/scripts/addons/phobos/blender/io/blender2phobos.py", line 818, in <listcomp>
    joints=[deriveJoint(obj) for obj in objectlist if obj.phobostype == 'link' and sUtils.getEffectiveParent(obj, include_hidden=True) is not None],
  File "~/.config/blender/3.6/scripts/addons/phobos/blender/utils/validation.py", line 806, in validation_wrapper
    return function(obj, *args, logging=logging, **kwargs)
  File "~/.config/blender/3.6/scripts/addons/phobos/blender/io/blender2phobos.py", line 474, in deriveJoint
    raise KeyError("joint/type: "+msg)
KeyError: 'joint/type: The joint type of visual_battery_holder_link has not been properly defined.'
```
I was following the Modeling Walkthrough and the Kinematic Skeleton page, and I finished making links, joints, and collisions. I tried to export and I got this error. It also doesn't change based on my export settings. I wanted to put a fixed joint between `visual_base_plate_link` and `visual_battery_holder_link`, but nothing visually changes when I do, no visible properties change, and it still has the same issue with the same link.

**Expected behavior**
I expected `joint/type` to be defined when I used the `Define Joint(s)` operator.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Ubuntu 22.04.3 LTS
 - Blender version: 3.6.3 (but this also happened on a prev)

**Additional context**
Add any other context about the problem here.

hwiedPro, 2023-09-26:
Hi @literally-anything ,
It's hard to tell what went wrong at your end without inspecting your model, do you mind sharing your blend file?
I just tested the Define Joint operator and for me it works fine.

literally-anything, 2023-10-01:
Sorry for the late response. Here is my blend file: [avr_phobos.tar.gz](https://github.com/dfki-ric/phobos/files/12776192/avr_phobos.tar.gz)

I tried using `Define Joint(s)` without the root link selected and it still would not export.

svepe, 2023-10-26:
Not sure if it is related, but I've noticed that if some components are hidden from the blender scene, then all sorts of errors such as this one happen.

hwiedPro, 2024-02-13:
@svepe You are right blender only applies to not hidden objects.

@literally-anything Sorry I currently don't have time to give detailed support on your model. Maybe @AlpenAalAlex can have a look?

AlpenAalAlex, 2024-03-05:
@literally-anything Sorry late answer. I assume this is a bug in Blender. I noticed your links are in local view and are missing in the list of selected objects. I left local view and defining joints worked fine. Please tell me if this helps.

[image]