dfki-ric/phobos #338: Exporting scaled meshes

23pointsNorth, 2023-10-12:
**Describe the bug**
I've sketched up a scene and rigged it with motors/joints. 
(1) When exporting without [Apply mesh scale], the process succeeds, but with wrong model, as some of the meshes have non-uniform x,y,z, scales + the collision boxes don't get loaded in gazebo.
(2) When exporting with selected checkbox on [Apply mesh scale], I am getting the following error:

```
Error: Python: Traceback (most recent call last):
  File "/home/bot/.config/blender/3.3/scripts/addons/phobos/blender/operators/io.py", line 142, in invoke
    return self.execute(context)
  File "/home/bot/.config/blender/3.3/scripts/addons/phobos/blender/operators/io.py", line 260, in execute
    self.exportModel(root, exportpath)
  File "/home/bot/.config/blender/3.3/scripts/addons/phobos/blender/operators/io.py", line 215, in exportModel
    robot.export(
  File "/home/bot/.config/blender/3.3/scripts/addons/phobos/core/robot.py", line 682, in export
    self.export_meshes(mesh_output_dir=os.path.join(outputdir, rel_mesh_pathes[mf]), format=mf, apply_scale=apply_scale)
  File "/home/bot/.config/blender/3.3/scripts/addons/phobos/core/robot.py", line 61, in export_meshes
    vc.geometry.provide_mesh_file(targetpath=os.path.abspath(mesh_output_dir), format=format,
  File "/home/bot/.config/blender/3.3/scripts/addons/phobos/io/representation.py", line 916, in provide_mesh_file
    self.apply_scale()
  File "/home/bot/.config/blender/3.3/scripts/addons/phobos/io/representation.py", line 1113, in apply_scale
    bpy.ops.object.transform_apply(scale=True)
  File "/snap/blender/4022/3.3/scripts/modules/bpy/ops.py", line 113, in __call__
    ret = _op_call(self.idname_py(), None, kw)
RuntimeError: Error: Cannot apply to a multi user: Object "tmp_export_mesh.001", Mesh "mesh.001", aborting
```
Attempted to select the Object -> Relations -> Make Single User -> Object & Data & Materials, but it doesn't change the result. Is this Phobos related, or blender related?

**To Reproduce**
Steps to reproduce the behavior:
1. Take model. 
[excavator_dummy_boom.zip](https://github.com/dfki-ric/phobos/files/12880080/excavator_dummy_boom.zip)
2. Click on export model with [Apply mesh scale] selected.
3. See error

**Expected behavior**
For the exported `dae` meshes to contain the correct scale. 

**Desktop (please complete the following information):**
 - OS: [e.g. Ubuntu] Version 22.04]
 - Blender version 3.3

**Additional context**
(If they sound like bug and not user errors, can create issues)
1. Another interesting quirk -> if a joint is rotated by 90 deg, the collision body generated from Phobos of a box, approximated with a box is very off (create a collision for visual object `boom` in this model.)
2. Should all meshes be able to generate an inertial object? 
```
Original exception was:
File "/home/bot/.config/blender/3.3/scripts/addons/phobos/blender/operators/editing.py", line 1048, in toggleVisual
Traceback (most recent call last):
  File "/home/bot/.config/blender/3.3/scripts/addons/phobos/blender/operators/editing.py", line 1189, in execute
    inertia = inertialib.calculateInertia(obj, mass, geometry, adjust=True, logging=True)
  File "/home/bot/.config/blender/3.3/scripts/addons/phobos/blender/utils/validation.py", line 806, in validation_wrapper
    return function(obj, *args, logging=logging, **kwargs)
  File "/home/bot/.config/blender/3.3/scripts/addons/phobos/blender/model/inertia.py", line 70, in calculateInertia
    inertia = calculateMeshInertia(mass, obj.data, scale=obj.scale)
  File "/home/bot/.config/blender/3.3/scripts/addons/phobos/utils/inertia.py", line 150, in calculateMeshInertia
    triangles += [face]
ValueError: operands could not be broadcast together with shapes (0,) (1,17) (0,) 
Error: Python: Traceback (most recent call last):
  File "/home/bot/.config/blender/3.3/scripts/addons/phobos/blender/operators/editing.py", line 1189, in execute
    inertia = inertialib.calculateInertia(obj, mass, geometry, adjust=True, logging=True)
  File "/home/bot/.config/blender/3.3/scripts/addons/phobos/blender/utils/validation.py", line 806, in validation_wrapper
    return function(obj, *args, logging=logging, **kwargs)
  File "/home/bot/.config/blender/3.3/scripts/addons/phobos/blender/model/inertia.py", line 70, in calculateInertia
    inertia = calculateMeshInertia(mass, obj.data, scale=obj.scale)
  File "/home/bot/.config/blender/3.3/scripts/addons/phobos/utils/inertia.py", line 150, in calculateMeshInertia
    triangles += [face]
ValueError: operands could not be broadcast together with shapes (0,) (1,17) (0,) 
```

hwiedPro, 2024-06-20:
@23pointsNorth Did you try @AlpenAalAlex PR?

23pointsNorth, 2024-06-20:
Sorry, I have not. I can see the model is attached in the original comment, so potentially can be tested.

commit fb867321 (2025-03-25): Fix #338 (2)
closed 2025-03-25 by AlpenAalAlex