dfki-ric/phobos #340: Cannot activate addon: ModuleNotFoundError: No module named 'pkg_resources'

ilmagico, 2024-01-11:
**Describe the bug**
I installed blender 3.3LTS via snap on manjaro linux (`sudo snap install blender --channel=3.3lts/stable --classic`), then I downloaded phobos 2.0.0 from [here](https://github.com/dfki-ric/phobos/releases/download/2.0.0/phobos.zip).
Started blender, went into preferences -> addons, installed from file, and when I tried to enable it, it spent some time installing its dependencies and eventually failed with this error:

```
Checking requirements:
Looking in links: /tmp/tmpogrcc3ud
Requirement already satisfied: setuptools in ./.local/lib/python3.10/site-packages (65.5.0)
Requirement already satisfied: pip in ./.local/lib/python3.10/site-packages (23.3.2)
  Upgrading pip...
Defaulting to user installation because normal site-packages is not writeable
Requirement already satisfied: pip in ./.local/lib/python3.10/site-packages (23.3.2)
  Checking yaml
  Checking numpy
  Checking scipy
  Checking collada
  Checking pydot
  Checking lxml
  Checking networkx
  Checking trimesh
  Checking PIL
Phobos:All Phobos requirements have been installed.
Please restart Blender to activate the Phobos add-on!
Exception in module register(): /home/ilmagico/.config/blender/3.3/scripts/addons/phobos/__init__.py
Traceback (most recent call last):
  File "/home/ilmagico/.config/blender/3.3/scripts/addons/phobos/__init__.py", line 143, in register
    from . import io
  File "/home/ilmagico/.config/blender/3.3/scripts/addons/phobos/io/__init__.py", line 1, in <module>
    from . import representation
  File "/home/ilmagico/.config/blender/3.3/scripts/addons/phobos/io/representation.py", line 14, in <module>
    from .xml_factory import singular as _singular, plural as _plural
  File "/home/ilmagico/.config/blender/3.3/scripts/addons/phobos/io/xml_factory.py", line 10, in <module>
    import pkg_resources
ModuleNotFoundError: No module named 'pkg_resources'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/var/lib/snapd/snap/blender/4325/3.3/scripts/modules/addon_utils.py", line 369, in enable
    mod.register()
  File "/home/ilmagico/.config/blender/3.3/scripts/addons/phobos/__init__.py", line 213, in register
    raise ImportWarning(installation_finished_message)
ImportWarning: All Phobos requirements have been installed.
Please restart Blender to activate the Phobos add-on!
```

**To Reproduce**
Steps to reproduce the behavior:
1. Install blender 3.3LTS (via snap in my case)
2. Download Phobos 2.0.0
3. Install downloaded .zip addon into blender
4. Try to enable the addon

**Expected behavior**
The addon is enabled and works.

Note: the very first time, it also gave an error about `trimesh`, which was replaced by the one above after restarting blender. It might or might not be related:
```
Installing collected packages: numpy, trimesh
Successfully installed numpy-1.26.3 trimesh-4.0.8
  Checking PIL
  Installing package Pillow
Collecting Pillow
  Downloading pillow-10.2.0-cp310-cp310-manylinux_2_28_x86_64.whl.metadata (9.7 kB)
Downloading pillow-10.2.0-cp310-cp310-manylinux_2_28_x86_64.whl (4.5 MB)
   ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 4.5/4.5 MB 46.5 MB/s eta 0:00:00
Installing collected packages: Pillow
Successfully installed Pillow-10.2.0
Phobos:All Phobos requirements have been installed.
Please restart Blender to activate the Phobos add-on!
Exception in module register(): /home/ilmagico/.config/blender/3.3/scripts/addons/phobos/__init__.py
Traceback (most recent call last):
  File "/home/ilmagico/.config/blender/3.3/scripts/addons/phobos/__init__.py", line 143, in register
    from . import io
  File "/home/ilmagico/.config/blender/3.3/scripts/addons/phobos/io/__init__.py", line 1, in <module>
    from . import representation
  File "/home/ilmagico/.config/blender/3.3/scripts/addons/phobos/io/representation.py", line 9, in <module>
    import trimesh
ModuleNotFoundError: No module named 'trimesh'
```

**Desktop (please complete the following information):**
 - OS: Manjaro (rolling distro, just updated)
 - Blender version 3.3.14
 - debugging output (run `setup.py --info`): not sure where to find this

ilmagico, 2024-01-12:
I also just tried with the latest `master` branch straight from git, and I get the same error.

AlpenAalAlex, 2024-01-15:
@ilmagico Please checkout the commit I just pushed and see if this fixes your issue

ilmagico, 2024-01-15:
@AlpenAalAlex Yep, that fixes it, thank you!!! I guess that should become a PR :)

closed 2024-01-31 by AlpenAalAlex